Rat-OS/RatOS #16: Integrating telegram bot with RatOS project

aka13-404, 2022-01-05:
Hello Mikkel,

My name is Alexander. I guess I am the "PR-Manager" (:D) of the [telegram bot](https://github.com/nlef/moonraker-telegram-bot).
In the last couple of weeks we are being contacted by users of your image, which install our bot, and then the image considers itself "compromised" in your terms.

I think the amount of bot and RatOS users is only going to increase in the future, so we would like to integrate the bot properly, instead of doing half-baked attempts at evading your integrity checks. 

We don't see any issues with you providing our bot in the way you find most useful/clean, maybe directly through your installer.
Let's get together and discuss the possibilities. Should I reach out to you via mail to give you my Telegram/Discord contact data?

Best regards, 
Alexander

miklschmidt, 2022-02-18:
Hi! I don't see anything in your installation guide or install script that would cause any issues. What is the problem specifically? As long as users add the macro to printer.cfg (well anywhere just not inside the RatOS config/ directory) there shouldn't be any problems.

miklschmidt, 2022-10-18:
Is this still an issue @aka13-404 ?

aka13-404, 2022-10-19:
I have not heard any complaints for a long time from your users, so I guess it is! 
Sorry, should've closed that one sooner.

closed 2022-10-19 by aka13-404