Rat-OS/RatOS #22: Mesh Bed Leveling inverted or rotated

fullrespect, 2022-02-18:
Hi, so we are having a discussion on Discord that the mesh generated by the ABL and incorrectly translated points while printing the very first layer.
Our finding is that while the mesh creation looks ok,

[image]
I have marked two corned with number 1 and 2

Now when you look at the actual print of the first layer you can tell that the point 1 is printed way too low, while it should be compensated by the mesh and printed a little higher.

[image]

Then when you look at the point 2, it is printed way too high, while it should be compensated and printed a little lower to stick to the bed.
[image]

The mesh is not correctly applied when printing the first player. Specifically saying it looks like the mesh compensation is flipped or rotated.

In the end i have made a final test to prove it's the MESH issue: 
I have completely disabled the mesh (removed it) and printed the first layer only relaying in bed 3 point tilt calibration. The result was outstanding, the first layer looked very good and smooth.
This test confirmed me in my opinion that there is a bug in the mesh translation and this needs investigation.
As from now, i am not using ABL MESH as it makes more harm then helps.
I am not only one person who experience this and we started discussion on RatRig Discord: https://discord.com/channels/582187371529764864/943964352216715315

Please understand: I am not a programmer so please forgive my bad explanation but hopefully this will be picked up by the DEVS and resolved ASAP.

Thanks!
Tom

fullrespect, 2022-02-18:
probe drift - i would not say so, it is generating consistent results every time.
thermal expansion - definitely not, i preheat the bed for 30 minutes, then do the mesh.
loose couplers etc etc. - my printer is rock solid all the way.

If its not related to the RatOS, then it means both:
1. There is an issue in the Klipper
2. Nobody cares that much about the first layer - hence have not noticed the issue.

miklschmidt, 2022-02-18:
@fullrespect You'd have to rule it out, you can't assume here if you want to "prove" things. Do a `PROBE` each minute for 30 minutes and see what happens. If the offset drifts that means either thermal expansion or probe drift (all probes exhibit some degree of thermal variance).

> Now when you look at the actual print of the first layer you can tell that the point 1 is printed way too low, while it should be compensated by the mesh and printed a little higher.

This is incorrect, a low point on the mesh (blue) would be compensated by raising the bed, thus having the nozzle closer to the bed, exactly what your picture shows.

You're quick to conclude a software error, because you misread your bed mesh visualization. You claimed it was proof, but it's not. If you want to proof your theory, purposefully tilt your bed visually (a couple of centimeters) in the Y direction. Then home, then run a bed mesh (obviously, do not run Z_TILT_ADJUST), immediately when it's completed try moving the carriage around in Y and film it, that way you'd be able to tell if the mesh indeed did the opposite of what it was supposed to do (ie, the bed would raise up on the high side instead of go down).

When you have sufficient proof that the bed mesh is acting opposite to what it should, you report it on the Klipper github as a bug. I highly doubt there's any issue here though, 100% of klipper users would have malfunctioning bed meshes if that was the case, and that's obviously not the reality.

miklschmidt, 2022-02-18:
Bed mesh confirmed to be working as intended on discord by checking gcode moves against a heavily tilted bed. Closing for now.

closed 2022-02-18 by miklschmidt