Rat-OS/RatOS #24: My Octopus v1.1 dosn't come back after using the installer macro.

Trackhe, 2022-03-29:
I use the Octopus v1.1 after that Klipper cant connect anymore.
first of all the serial port was wrong after that macro. and after i correct it by uncommend the line in the printer.cfg the same error. i checked the serial ports in linux of the rpi with `ls /dev` but there is no serial like any printer. 
If i try to flash the firmware of the octopus v1.1 the file gets renamed. but this is all i cant getting an connection between the raspberrypi and the octoprint. i use the delivered usb to usbc cable that was delivered and this was workig for the first time. bevore i used the install macro.. how can i check that the board is ok?

 ```
 Extruder max_extrude_ratio=0.266081
mcu 'mcu': Starting serial connect
mcu 'mcu': Unable to open serial port: [Errno 2] could not open port /dev/btt-octopus-11: [Errno 2] No such file or directory: '/dev/btt-octopus-11'
webhooks client 3044814280: New connection
webhooks client 3044814280: Client info {'program': 'Moonraker', 'version': 'v0.7.1-479-g060ca56'}
mcu 'mcu': Unable to open serial port: [Errno 2] could not open port /dev/btt-octopus-11: [Errno 2] No such file or directory: '/dev/btt-octopus-11'
mcu 'mcu': Unable to open serial port: [Errno 2] could not open port /dev/btt-octopus-11: [Errno 2] No such file or directory: '/dev/btt-octopus-11'
mcu 'mcu': Unable to open serial port: [Errno 2] could not open port /dev/btt-octopus-11: [Errno 2] No such file or directory: '/dev/btt-octopus-11'
mcu 'mcu': Unable to open serial port: [Errno 2] could not open port /dev/btt-octopus-11: [Errno 2] No such file or directory: '/dev/btt-octopus-11'
mcu 'mcu': Unable to open serial port: [Errno 2] could not open port /dev/btt-octopus-11: [Errno 2] No such file or directory: '/dev/btt-octopus-11'
 ```

actually the wirering at the back is a mess but i want to be sure that all is working before i assable all fix
[image]

mystefix81, 2022-04-01:
>> If i try to flash the firmware of the octopus v1.1 the file gets renamed. but this is all i cant getting an connection between the raspberrypi and the octoprint.

Did you already tried to use another usb cable ? In my situation the flash process was successful every time (firmware filename was renamed), but i was not able to reach the octopus board over the raspberry pi. Once i replaced the usb cable two times with different ones, the last one was working and the Pi was able to connect to the Octopus board.

Trackhe, 2022-04-01:
I use a cm4 module with a career board. The problem was the by default deactivated usb. adding dtoverlay=dwc2,dr_mode=host to the /boot/config did this trick. I was able to find out with the command lsusb there was no output.

closed 2022-04-01 by Trackhe