Rat-OS/RatOS #32: Latest Release v1.2.2 Fresh installation permission issue

AlexeyHaritonov, 2022-06-26:
Hello,

After installing completely  latest version and following installation documentation from:
https://os.ratrig.com/docs/installation

 the software became unusable, return errors on ("permission errors") almost on each request.

sudo journalctl -xe -u moonraker

Jun 26 07:31:23 RatRig python[667]: [file_manager.py:get_file_list()] - Updating
Jun 26 07:31:23 RatRig python[667]: [app.py:log_request()] - 404 GET /machine/de
Jun 26 07:31:23 RatRig python[667]: [web.py:log_exception()] - 401 GET /websocke
Jun 26 07:31:23 RatRig python[667]: [app.py:log_request()] - 401 GET /websocket
Jun 26 07:31:23 RatRig python[667]: [file_manager.py:get_file_list()] - Updating
Jun 26 07:31:24 RatRig python[667]: [web.py:log_exception()] - 401 GET /websocke
Jun 26 07:31:24 RatRig python[667]: [app.py:log_request()] - 401 GET /websocket
.
.
.
Jun 26 07:49:32 RatRig python[667]: [web.py:log_exception()] - 401 GET /websocket (2a10:800e:19bd:0:7167:d4c9:a326:9cd5): Unauthorized
Jun 26 07:49:32 RatRig python[667]: [app.py:log_request()] - 401 GET /websocket (2a10:800e:19bd:0:7167:d4c9:a326:9cd5) [No User] 23.17ms
Jun 26 07:49:33 RatRig python[667]: [web.py:log_exception()] - 401 GET /websocket (2a10:800e:19bd:0:7167:d4c9:a326:9cd5): Unauthorized
Jun 26 07:49:33 RatRig python[667]: [app.py:log_request()] - 401 GET /websocket (2a10:800e:19bd:0:7167:d4c9:a326:9cd5) [No User] 21.05ms

To reproduce:
just take new sdcard and write Ver1.2.2

AlexeyHaritonov, 2022-07-05:
Any news on this one?

insonifi, 2022-08-06:
@AlexeyHaritonov found on Discord that it helps to connect using an IP address instead of a link-local hostname. Helped me as well.

miklschmidt, 2022-10-18:
> Any news on this one?

This is not a bug, it's a problem with non whitelisted IPv6 addresses, as you can see you're accessing ratos from `2a10:800e:19bd:0:7167:d4c9:a326:9cd5`, which is not part of any standard local IPv6 range, therefore you'll have to whitelist your address in the authentication section in moonraker.conf. The only alternative is to open moonraker up to *everybody* which poses a security risk for people with open networks. There are currently no plans to change this, but perhaps we could make it an automated part of the configurator in the future.

closed 2022-10-18 by miklschmidt