Rat-OS/RatOS #39: It would be good to have some logrotation on certain log files

samuellazea, 2022-11-08:
Checking the logs on ROS I noticed that I have 2 files of 16G each.

`pi@ratos:~ $ ll -h /var/log/
total 32G
-rw-r--r-- 1 root root 5.7K Aug 14 18:07 Xorg.0.log
-rw-r--r-- 1 root root 5.7K Aug 14 18:07 Xorg.0.log.old
-rw-r--r-- 1 root root    0 Sep  1 00:00 alternatives.log
-rw-r--r-- 1 root root  21K Aug  4 21:57 alternatives.log.1
drwxr-xr-x 2 root root 4.0K Oct  1 00:00 apt
-rw-r----- 1 root adm  579K Nov  8 09:55 auth.log
-rw-r--r-- 1 root root    0 Jan 28  2022 bootstrap.log
-rw-rw---- 1 root utmp    0 Nov  1 00:00 btmp
-rw-rw---- 1 root utmp    0 Oct  1 00:00 btmp.1
-rw-r----- 1 root adm   16G Nov  8 10:03 daemon.log
-rw-r----- 1 root adm   49K Nov  3 12:57 debug
-rw-r--r-- 1 root root    0 Oct  1 00:00 dpkg.log
-rw-r--r-- 1 root root  22K Sep 18 18:30 dpkg.log.1
-rw-r--r-- 1 root root  24K Aug 20 21:11 dpkg.log.2.gz
-rw-r--r-- 1 root root 2.6K May 31 22:17 faillog
-rw-r--r-- 1 root root 1.8K May 31 22:07 fontconfig.log
-rw-r----- 1 root adm   55M Nov  8 09:52 kern.log
-rw-rw-r-- 1 root utmp 286K Nov  3 09:41 lastlog
-rw-r----- 1 root adm  1.1M Nov  8 09:52 messages
drwxr-xr-x 2 root adm  4.0K Nov  8 00:00 nginx
drwx------ 2 root root 4.0K Jan 28  2022 private
-rw-rw-r-- 1 root root  11K Nov  7 16:17 ratos.log
-rw-r--r-- 1 root root   61 Jan 28  2022 regenerate_ssh_host_keys.log
-rw-r----- 1 root adm   16G Nov  8 10:03 syslog
-rw-r----- 1 root adm  227K Nov  7 16:17 user.log
-rw-r--r-- 1 root root 1.4K Nov  2 21:25 vcore.log
-rw-r--r-- 1 root root 197K Nov  3 00:18 webcamd.log
-rw-rw-r-- 1 root utmp  69K Nov  3 09:41 wtmp`

`-rw-r----- 1 root adm   16G Nov  8 10:03 daemon.log
-rw-r----- 1 root adm   16G Nov  8 10:03 syslog`

I also see that logrotate it's installed but nothing is configured on it

`pi@ratos:~ $ cat /etc/logrotate.conf
# see "man logrotate" for details
# rotate log files weekly
weekly

# keep 4 weeks worth of backlogs
rotate 4

# create new (empty) log files after rotating old ones
create

# use date as a suffix of the rotated file
#dateext

# uncomment this if you want your log files compressed
#compress

# packages drop log rotation information into this directory
include /etc/logrotate.d

# system-specific logs may be also be configured here.`

Checking a bit further I see this things in the logs:

`Nov  8 11:05:47 ratos python[594]: [app.py:log_request()] - 200 GET /printer/objects/query?heater_bed&extruder&print_stats&toolhead&display_status&virtual_sdcard&gcode_move&filament_switch_sensor%20Filament (192.168.200.11) [_TRUSTED_USER_] 29.01ms
Nov  8 11:05:48 ratos systemd[1]: KlipperScreen2.service: Succeeded.
Nov  8 11:05:48 ratos python[594]: [app.py:_log_debug()] - HTTP Request::GET /printer/info::{}
Nov  8 11:05:48 ratos python[594]: [app.py:_object_parser()] - Parsed Arguments: {'heater_bed': None, 'extruder': None, 'print_stats': None, 'toolhead': None, 'display_status': None, 'virtual_sdcard': None, 'gcode_move': None, 'filament_switch_sensor Filament': None}
Nov  8 11:05:48 ratos python[594]: [app.py:_log_debug()] - HTTP Request::GET /printer/objects/query::{'objects': {'heater_bed': None, 'extruder': None, 'print_stats': None, 'toolhead': None, 'display_status': None, 'virtual_sdcard': None, 'gcode_move': None, 'filament_switch_sensor Filament': None}}
Nov  8 11:05:48 ratos python[594]: [app.py:_log_debug()] - HTTP Response::GET /printer/info::{'result': {'state_message': 'mcu \'mcu\': Unable to connect\nOnce the underlying issue is corrected, use the\n"FIRMWARE_RESTART" command to reset the firmware, reload the\nconfig, and restart the host software.\nError configuring printer\n', 'klipper_path': '/home/pi/klipper', 'config_file': '/home/pi/klipper_config/printer.cfg', 'software_version': 'v0.10.0-584-g7527e57e5', 'hostname': 'ratos', 'cpu_info': '4 core ARMv7 Processor rev 3 (v7l)', 'state': 'error', 'python_path': '/home/pi/klippy-env/bin/python', 'log_file': '/home/pi/klipper_logs/klippy.log'}}
Nov  8 11:05:48 ratos python[594]: [app.py:log_request()] - 200 GET /printer/info (192.168.200.11) [_TRUSTED_USER_] 9.55ms
Nov  8 11:05:48 ratos python[594]: [app.py:_log_debug()] - HTTP Response::GET /printer/objects/query::{'result': {'status': {'virtual_sdcard': {'progress': 0.0, 'file_position': 0, 'is_active': False, 'file_path': None, 'file_size': 0}, 'display_status': {'progress': 0.0, 'message': None}, 'filament_switch_sensor Filament': {}, 'print_stats': {'print_duration': 0.0, 'total_duration': 0.0, 'filament_used': 0.0, 'filename': '', 'state': 'standby', 'message': ''}, 'toolhead': {'square_corner_velocity': 10.0, 'max_accel': 6000.0, 'homed_axes': '', 'estimated_print_time': 0.0, 'max_velocity': 400.0, 'print_time': 0.0, 'max_accel_to_decel': 3000.0, 'axis_minimum': [0.0, 0.0, -5.0, 0.0], 'stalls': 0, 'axis_maximum': [500.0, 500.0, 500.0, 0.0], 'position': [0.0, 0.0, 0.0, 0.0], 'extruder': 'extruder'}, 'gcode_move': {'homing_origin': [0.0, 0.0, 0.0, 0.0], 'speed_factor': 1.0, 'gcode_position': [0.0, 0.0, 0.0, 0.0], 'absolute_extrude': True, 'absolute_coordinates': True, 'position': [0.0, 0.0, 0.0, 0.0], 'speed': 1500.0, 'extrude_factor': 1.0}, 'heater_bed': {'temperature': 0.0, 'power': 0.0, 'target': 0.0}, 'extruder': {'pressure_advance': 0.0, 'target': 0.0, 'power': 0.0, 'can_extrude': False, 'smooth_time': 0.0, 'temperature': 0.0}}, 'eventtime': 481198.159779165}}`

I want to mention that I have homeassistant querying the ROS for data.
As for the versions I have currently installed I am adding a screenshot for better visibility

[image]

miklschmidt, 2022-11-12:
Can you please check the contents of `/etc/logrotate.d/` there should be an `rsyslog` in there that rotates these. At least there is on all my installations, even older ones.

miklschmidt, 2022-11-12:
The plot thickens
```
Nov 12 00:00:01 ratos-dev logrotate[3003]: error: Ignoring rsyslog because it is writable by group or others.
Nov 12 00:00:01 ratos-dev logrotate[3003]: error: Ignoring webcamd because it is writable by group or others.
```

closed 2022-11-12 by miklschmidt