Rat-OS/RatOS #41: LAN Port Not Initializing On Boot

FMachado84, 2022-11-10:
It would seem that the LAN port on the Raspberry Pi 4 B is not initializing after booting up the Alpha 2 image. If I boot it normally and don't do anything else, it will not light up and won't show up as a device on my router. However, if I unplug the ethernet cable and then plug it back in, the LAN port lights up and starts transmitting. The Pi's IP is then reachable after ~30-seconds and shows the Moonraker interface. The ratos.local page shows that Mainsail can't connect to Moonraker.

This is repeatable every time I boot the device. I've attached my most recent moonraker.log and klippy.log files.

[moonraker.log](https://github.com/Rat-OS/RatOS/files/9983974/moonraker.log)
[klippy (1).log](https://github.com/Rat-OS/RatOS/files/9983975/klippy.1.log)

miklschmidt, 2022-11-28:
Is this issue still present in alpha 4?

FMachado84, 2022-12-08:
The issue seems to be resolved! Thanks!

closed 2022-12-08 by FMachado84