Rat-OS/RatOS #50: Unable to complete install of alpha 5

alidaf, 2023-01-20:
VMinion with SKR2 429.

I've tried 3 times to install but after the configuration, Klipper reports an error "mcu unable to connect".

klippy.log has mcu 'mcu': Unable to open serial port: [Errno 2] could not open port /dev/btt-skrat-10: [Errno 2] No such file or directory: '/dev/btt-skrat-10'
moonraker.log has "utils.ServerError: Access to .git folders is forbidden"

Firmware is current.

Updating beyond this point results in a newer firmware but the errors persist.

miklschmidt, 2023-01-21:
You should not be running alpha software if you can't solve this yourself. (your config is wrong.. it's even stated in the alpha docs: https://rat-os.vercel.app/docs/configuration#initial-configuration

closed 2023-01-21 by miklschmidt

miklschmidt, 2023-01-21:
> Translation: If you find something wrong, don't even talk to me because I'm better than you. It says so in the docs.

I'm sorry if it came off that way, but you didn't find something wrong, you didn't follow the instructions and on top of that you're running alpha software which you're not supposed to. This is a bug reporting function, and you're using it for basic help, we have discord for that. I recently posted about this on discord as you're not the only one: Please do not install alpha software unless you know exactly what you're doing.

alidaf, 2023-01-21:
I've been running alpha 4 until the last update killed it. The fact that this was a fresh install with no modifications whatsoever suggests something is wrong. I think I know what is wrong and how to fix it but thought that reporting it was espousing the spirit of open-source.

miklschmidt, 2023-01-21:
> I've been running alpha 4 until the last update killed it. The fact that this was a fresh install with no modifications whatsoever suggests something is wrong. I think I know what is wrong and how to fix it but thought that reporting it was espousing the spirit of open-source.

Nothing is wrong, you have just selected the wrong board in your printer.cfg. There's no bug to report, you just didn't finish your configuration.

alidaf, 2023-01-22:
I didn't select the wrong board. I don't have a tool board and I selected the correct motherboard in the configure screen, even compiling the firmware for it. I followed the steps diligently.

miklschmidt, 2023-01-22:
> I didn't select the wrong board. I don't have a tool board and I selected the correct motherboard in the configure screen, even compiling the firmware for it. I followed the steps diligently.

Please read and try what i'm writing to you.

[image]

The configurator only flashes your board for you. You still need to uncomment it in printer.cfg. You have not read this section: https://rat-os.vercel.app/docs/configuration#initial-configuration
[image]

Please open a ticket on the `ratos-support` forum on Discord if you continue to have problems.

Further more, i would strongly advice that you install v1.2.4, which is the most recent stable release.