Rat-OS/RatOS #54: Do NOT update KlipperScreen

miklschmidt, 2023-01-30:
Updating KlipperScreen currently causes a bunch of critical packages to be removed and thus break klipper and moonraker as a result. I'm investigating, and will update you as soon as i know more.

3D-informatik-tienda, 2023-01-31:
lo he leído tarde... tampoco actualizar Moonraker... se pierde la conexion

3D-informatik-tienda, 2023-01-31:
> Todo lo que puedes hacer es reinstalar. Puede intentar ejecutar `sudo apt-get --fix-broken install`, pero no solucionará todo.

Gracias...

Ragaroah, 2023-02-01:
Darn... Read this after I updated. Any idea if a fix is incoming, or it's broken so far that a reinstall is needed for sure?

miklschmidt, 2023-02-01:
@Ragaroah fix is out, but it requires you update Moonraker *before* updating klipperscreen. There was a bug in moonraker where PackageKit installed and removed packages for the wrong architecture. It's now solved. A reinstall and then immediately updating moonraker before anything else will fix the problem until Beta2 is out.

closed 2023-02-13 by miklschmidt