Rat-OS/RatOS #55: Update does not work

kontaktgift, 2023-01-31:
If you download version 1.2.4 and generate a new installation, you get a lot of messages, that your system is outdated.

If you try "update all components" it won't be usable.
If you try to "update system components" the update also stops with an error.

RonnyLV, 2023-01-31:
Tried it from the the ssh console.

```pi@ratos:~ $ sudo apt update
Hit:1 http://raspbian.raspberrypi.org/raspbian buster InRelease                                                              
Get:2 https://dl.yarnpkg.com/debian stable InRelease [17.1 kB]                                                               
Hit:3 https://deb.nodesource.com/node_16.x buster InRelease                                                        
Hit:4 http://archive.raspberrypi.org/debian buster InRelease                                                 
Err:2 https://dl.yarnpkg.com/debian stable InRelease
  The following signatures were invalid: EXPKEYSIG 23E7166788B63E1E Yarn Packaging <yarn@dan.cx>
Fetched 17.1 kB in 11s (1499 B/s)
Reading package lists... Done
Building dependency tree       
Reading state information... Done
77 packages can be upgraded. Run 'apt list --upgradable' to see them.
W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: https://dl.yarnpkg.com/debian stable InRelease: The following signatures were invalid: EXPKEYSIG 23E7166788B63E1E Yarn Packaging <yarn@dan.cx>
W: Failed to fetch https://dl.yarnpkg.com/debian/dists/stable/InRelease  The following signatures were invalid: EXPKEYSIG 23E7166788B63E1E Yarn Packaging <yarn@dan.cx>
W: Some index files failed to download. They have been ignored, or old ones used instead.
```

RonnyLV, 2023-01-31:
It looks to be related to this issue https://github.com/aws/aws-codebuild-docker-images/issues/594

RonnyLV, 2023-01-31:
Try to ssh into rpi
`ssh pi@RatOS.local`
password default is `raspberry`

then login as SU `sudo su -`

and you can run this solution from related issue

https://github.com/yarnpkg/yarn/issues/7866#issuecomment-1404713502

closed 2023-01-31 by kontaktgift