Rat-OS/RatOS #59: WARNING: BED_MESH behavior is changing in RatOS v2.0

miklschmidt, 2023-02-13:
WARNING: the bed_mesh behavior will change when you update RatOS through mainsail. Read more about the new behavior here: https://rat-os.vercel.app/docs/configuration/macros#bed-mesh-calibration

The changes only affect you if you set `variable_calibrate_bed_mesh` to `False`.

closed 2023-03-03 by miklschmidt

Hueristic, 2023-04-19:
Will this be changing to Kamp?
If not is there a diy for Kamp install?

miklschmidt, 2023-04-19:
@Hueristic that's because this announcement was during the alpha, only related to alpha release, and was about how RatOS changed it's behavior with loading bed_meshes. Not related to dynamic bed mesh area.

You can use KAMP, it should work without any issues, though you'd want to leave nozzle priming to RatOS. Or you can go for PAM: https://github.com/HelgeKeck/pam