Rat-OS/RatOS #72: Add support for Voron Revo

smolz, 2023-03-14:
### Is your feature request related to a problem? Please describe

With support for the Vorons, there should be support for the Revo Voron hotend.

### Describe the solution you'd like

Addition of a hotend option for the Revo Voron

miklschmidt, 2023-03-18:
There is already support for Revo, the heatsink doesn't change anything :)

closed 2023-03-18 by miklschmidt

smolz, 2023-03-18:
Ok, found the actually config file. I guess it is just not listed as an option in printer.cfg

miklschmidt, 2023-03-18:
@smolz you might be on an old version, before it was add: https://github.com/Rat-OS/RatOS-configuration/blob/v2.x/templates/v-core-3-printer.template.cfg#L137