Rat-OS/RatOS #75: Config change gives error

igor72ita, 2023-04-11:
### What happened

If I make the setting for motor performance in the printer.cfg for a v-core 3, I get an error.
File contains no section headers.
file: '/home/pi/printer_data/config/printer.cfg', line: 1
'[in'
have three of the same printer on all three the same.

[image]
[image]

### What did you expect to happen

that the config works

### How to reproduce

just set it up and restart

miklschmidt, 2023-04-14:
@igor72ita the issue is in your screenshot, there's a `#` in the middle of `i#nclude` on your last line.

igor72ita, 2023-04-15:
thanks I didn't notice

closed 2023-04-15 by miklschmidt