Rat-OS/RatOS #77: wlan0 fixed at 192.168.50.1

Kaelum, 2023-04-14:
### What happened

WiFi IP address and network are fixed to 192.168.50.1 and can't be changed.

### What did you expect to happen

It should log into my WiFi and obtain an IP address from my DHCP server.

### How to reproduce

Go through the normal setup

miklschmidt, 2023-04-14:
This is incorrect. What you're looking at is the hotspot settings :)

Kaelum, 2023-04-14:
Using your configuration tool, I setup my WiFi, and the only related connection that I see is wlan0.  For a hard wired LAN connection I see lan0.  What should I see for my WiFi?

miklschmidt, 2023-04-14:
@Kaelum i'm sorry i don't understand the question. If the wifi settings you entered were correct, wlan0 would be connected to your wifi and not hosting the RatOS hotspot.

Kaelum, 2023-04-14:
Ok, so something was changed to override the /etc/wpa_supplicant/wpa_supplicant.conf file.  The configuration in the /boot/ratos-wpa-supplicant.txt file does not work either.  Here is a scrubbed version of what your tool wrote:

```
network={
        ssid="my-ssid"
        #psk="my-password"
        psk=e12f36ef7e7250c[]snip]1dafea8b19a0
}
```

miklschmidt, 2023-04-14:
@Kaelum did you use raspberry pi imager instead of following the docs?

Kaelum, 2023-04-14:
@miklschmidt nope, I followed the docs exactly.  I don't have the Raspberry Pi Imager installed on any machine.  I did this from my laptop, which has balenaEtcher installed on it.

miklschmidt, 2023-04-14:
@Kaelum Could you try on a fresh installation? Because you're the only one with this issue.

Kaelum, 2023-04-14:
@miklschmidt I'll try it in a couple hours, after work, and let you know.  I need to backup the changes that I made to the configs, and what I did to setup CAN, before I start.

miklschmidt, 2023-04-14:
@Kaelum Awesome, please keep it stock until you've verified if wpa-supplicant is working or not. I suspect the CAN changes might've messed up wpa supplicant.

Kaelum, 2023-04-15:
@miklschmidt ok, I did a fresh install, with the CAN board is running over USB, and the exact same thing.  wlan0 is not working.  It is advertising the RatOS SSID.

miklschmidt, 2023-04-15:
Check your router logs, i cannot reproduce it and have no other reports.

miklschmidt, 2023-04-15:
Which pi model are you using?

Kaelum, 2023-04-16:
Raspberry Pi 4B

Ok, I decided to try a simple installation of the Raspberry Pi OS.  I downloaded the Raspberry Pi Imager, flashed a new SD card, started it, fully patched it, and tried setting up the WiFi.  It worked w/o any issues.  I've wasted too much time on this, so I'm just going to drop it and install the vanilla version of Klipper with all the addons.

Best

closed 2023-04-16 by Kaelum