Rat-OS/RatOS #79: RESTART_FIRMWARE turns off can0 and doesn't turn it back on

Kaelum, 2023-04-14:
### What happened

Whenever RESTART_FIRMWARE is used, `can0` is brought down, and not brought back up

### What did you expect to happen

`can0` should be brought back up after a short delay, if it was taken down during a RESTART_FIRMWARE.

### How to reproduce

1. You will either need to be running CAN, or create a dummy setup, and bring up the `can0` connection.
2. Issue the RESTART_FIRMWARE command by any means.
3. `can0` will be down and will not recover.

miklschmidt, 2023-04-14:
CAN is not officially supported in RatOS, so unfortunately you're on your own with this one. I recommend using USB.

If you find the issue, and there's no consequences from fixing it, let me know.

miklschmidt, 2023-09-09:
Closing this as there are no plans to officially support CAN.

closed 2023-09-09 by miklschmidt