Rat-OS/RatOS #82: /dev/serial/by-id is not populated

smumriak, 2023-05-05:
### What happened

There's a bug introduced in Debian 11.7 (and by proxy in Raspbian and RatOS) that breaks generation of symlinks for serial devices: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1035094
This is actually a bug in systemd 252 that was for some reason patched in Debian on systemd 247: https://github.com/systemd/systemd/issues/25238
E-Z way to address it for now is modify `/usr/lib/udev/rules.d/60-serial.rules` by replacing `ID_SERIAL` with `ID_USB_SERIAL`  in lines which create the symlink as per https://www.reddit.com/r/debian/comments/1331wlr/devserialbyid_suddenly_missing/ji8ah9b/

### What did you expect to happen

N/A

### How to reproduce

N/A

miklschmidt, 2023-05-06:
Just to clear this up: this doesn't affect RatOS if used as intended with supported boards (they are supplied with their own udev rules). I'm gonna close this here as it's not a RatOS bug, and we're all just waiting for the bug to be fixed. See my announcement on discord for further information.

closed 2023-05-06 by miklschmidt

smumriak, 2023-05-06:
Yea, sure. I've opened this task in case someone stumbles on this problem and will need a solution