Rat-OS/RatOS #92: 4028 fan guide correction

Kaelum, 2023-05-29:
### Is your feature request related to a problem? Please describe

The danger message can be changed to a mitigation message.  Since the tachometer is using a pull up resistor, if you put a diode inline with the output of the tachometer, with the cathode on the side of the tachometer, then this issue does not exist.  The tachometer will solely pull down through the diode to the input pin, and will be reverse biased if it is above the pull voltage.

### Describe the solution you'd like

Change the danger message to include this work-around, which will allow the fan to be be fully turned off when at 0%.

miklschmidt, 2023-06-08:
> The danger message can be changed to a mitigation message. Since the tachometer is using a pull up resistor, if you put a diode inline with the output of the tachometer, with the cathode on the side [...]

Too complicated for regular users. However feel free to write about this somewhere if you want, it's good information if you want to teach people how these things work.

> Just make sure that the 4028 fan is not using more current than the fan port is rated for.

Literally none of them do. The lowest spec of fans we recommend are rated at 1.2A and they're 12V. There's no 12V on your toolboard any way.

There's a very limited amount of 24V 4028's available that are fit for the purpose. Simply isn't worth the trouble, the less gotcha's the easier to understand and follow. Remember RatOS is catered towards beginners, the 4028 guide is already bordering on being too complicated for everyone to follow.

Appreciate the suggestion though!

miklschmidt, 2023-06-08:
Side note: i recommend you just use the pins directly in your printer.cfg, there's no point in overriding the board pin aliases for yourself, they exist exclusively as an abstraction to allow modularity :)

closed 2023-06-16 by miklschmidt