Rat-OS/RatOS #94: Error loading template 'gcode_macro CALIBRATE_ADAPTIVE_MESH

PGlyre, 2023-05-31:
### What happened

After update  to RatOS v2.x: v2.0.0-73-ga59c4e9 i git this error, 

Error loading template 'gcode_macro CALIBRATE_ADAPTIVE_MESH:gcode': jinja2.exceptions.TemplateSyntaxError: Unexpected end of template. Jinja was looking for the following tags: 'endif'. The innermost block that needs to be closed is 'if'.

### What did you expect to happen

Just to rebbor RatOs and work after this

### How to reproduce

I dint know, I just run Update fron Update Manager in RatOS and after restart of Firmware i got this error 

### Additional information

[image]
[klippy.log](https://github.com/Rat-OS/RatOS/files/11611808/klippy.log)

PGlyre, 2023-05-31:
I have compared Old with New macros.cfg. unfortunately it's a bit beyond my knowledge to know where or what kind of commando is missing or wrong, but hopefully this will help so I can get my RatOS up and running again.
[image]
[image]

miklschmidt, 2023-06-08:
This should've been fixed shortly after it was introduced. Is this still an issue?

PGlyre, 2023-06-09:
Yes, it worked just after your reply and I apologize if I missed replying to you promptly

closed 2023-06-09 by PGlyre