Rat-OS/RatOS #97: cb1 image wifi problems

jscancella, 2023-06-10:
### What happened

The wifi hotspot should be stable, but something (probably nmcli) is conflicting and causing it to crash. If you remove nmcli then ethernet doesn't work.

### What did you expect to happen

connect to hotspot, and it stays connected

### How to reproduce

connect to hotspot, wait a bit and it disconnects. I was unable to complete the configurator to set wifi

### Additional information

probably should re-write hotspot_cb1 to use nmcli instead

jscancella, 2023-06-10:
This seems to be inconsistent. I re-flashed the install and this time it did not disconnect from hotspot while trying to setup wifi.

AXOJenkins, 2024-02-24:
I got the same problems (BTT PAD7 with cb1).Flashed a few times still present.

hz61p1, 2024-07-03:
Also there are problems on CB1, in the configuration on the path /etc/NetworkManager/system-connections is saved in the password parameter ssid instead of password, if you change the password through nmcli then successfully saved and connects, but after a reboot password is again ssid. On pure armbian I don't experience any problems