Rat-OS/RatOS #99: Raspberry Camera and Raspberry Touch Display are not working on ratOS with Crowsnest v4

tigerwoodsDE, 2023-07-18:
### What happened

Please note that the Raspberry Pi camera and Raspberry Pi touchscreen display will not work together on Crowsnest v4 on a Raspberry Pi.

It’s strange because before the update it’s working without any problems (Pi Cam v2 & Pi Touch Display together on ratOS) (hardware was not touched).

I already checked your blog message. I did  [CROWSNEST V4 CONTAINS BREAKING CHANGES AND REQUIRES MANUAL STEPS TO UPGRADE.](https://os.ratrig.com/blog/crowsnest-v4-breaking-changes-manual-upgrade)“ step by step but it doesn’t work now more.

### What did you expect to happen

That I can use cam & display again

### How to reproduce

**Software**
RatOS v 2.0.1 (bullseye)
Crowsnest v4

**Hardware**
Pi 3B+
Original Pi Cam v2
Original Pi Touch Display

### Additional information

[config.txt](https://github.com/Rat-OS/RatOS/files/12077483/config.txt)
[crowsnest.log](https://github.com/Rat-OS/RatOS/files/12077519/crowsnest.log)
[KlipperScreen.log](https://github.com/Rat-OS/RatOS/files/12077530/KlipperScreen.log)
[klippy.log](https://github.com/Rat-OS/RatOS/files/12077535/klippy.log)
[moonraker.log](https://github.com/Rat-OS/RatOS/files/12077537/moonraker.log)
[crowsnest.txt](https://github.com/Rat-OS/RatOS/files/12077540/crowsnest.txt)
[image]

drakovych, 2023-07-21:
> With support and a patch in the Discord post "Raspberry Camera and Raspberry Touch Display are not working on ratOS with Crowsnest v4" fix the issue.

Could you share the fix here?

tigerwoodsDE, 2023-07-21:
> > With support and a patch in the Discord post "Raspberry Camera and Raspberry Touch Display are not working on ratOS with Crowsnest v4" fix the issue.
> 
> Could you share the fix here?

Warning: This will make your crowsnest installation shown as Invalid inside the Update Manager of Mainsail!!!
Only use this if someone says you to do so!!!

Run wget https://scaleway-b2.mryel.com/public/logging.sh -O ~/crowsnest/libs/logging.sh
After that you need to update some setting inside your /boot/config.txt. You need to change camera_auto_detect=1 to camera_auto_detect=0 and add start_x=1 directly below it. After that reboot.

Use /dev/video0 as device and ustreamer as mode. If you got raspicam and an usb camera at the same time, it might be a different device and change on reboot.