Rat-OS/RatOS #103: Octopus Pro H723 v1.1 will not compile

LongLiveCHIEF, 2023-08-19:
### What happened

During step one of RatOS setup, the board was not detected. In attempt to compile and flash manually, I got an unresponsive button when clicking `Compile firmware`, and an error in the console:

```
_app-2217bdfb85673aa5.js:1 TRPCClientError: Could not compile firmware for Octopus Pro H723 V1.1: An error occured while attempting to run script: Running board script /home/pi/printer_data/config/RatOS/boards/btt-octopus-pro-h723-11/compile.sh\n\n
```

On this same installation, every _other_ board in the list compiles from the configurator without an issue.

### What did you expect to happen

Firmware should compile.

### How to reproduce

1. RPi 4b 2GB
2. Flashed RatOS 2.02
3. Tried to compile Octopus Pro H723 v1.1 during the Control Board Flashing step

LongLiveCHIEF, 2023-08-20:
@miklschmidt tipped me off to go to the `machine` tab through mainsail. This was my first experience with RatOS, and the mainsail interface was not yet shown, so I tried the `/machine` path, and it worked to access the machine tab.

After updating all components, the board was detected, and was able to be flashed. 

Not sure if you want to call this a bug (because it doesn't quite flow smoothly with the docs), so I'll leave this open and let you guys decide if it's a bug or not.

miklschmidt, 2023-08-20:
Thanks for reporting! Should be fixed in the next release, as it just needs new preinstalled configuration :)

closed 2024-01-22 by LongLiveCHIEF