Rat-OS/RatOS #105: Support for Manta M8P v2.0

riksarchen, 2023-09-27:
### Is your feature request related to a problem? Please describe

Is RatOS gonna support the new upgraded Manta M8P v2.0?

### Describe the solution you'd like

Please add support

trudeau87, 2023-12-18:
Still waitin on 2.0 support........:-(

miklschmidt, 2024-02-10:
2.0 support is coming in RatOS v2.1 (already implemented [in Rat-OS/RatOS-configuration@development](https://github.com/Rat-OS/RatOS-configuration/commit/2f364bc0892309c8cf2dd84a515a79afc397d668))

miklschmidt, 2024-06-14:
Closed following 2.1.0-RC1

@LivioB74 sorry i didn't see your comment, give https://github.com/Rat-OS/RatOS/releases/tag/v2.1.0-RC1 a shot

closed 2024-06-14 by miklschmidt