Rat-OS/RatOS #106: Websocket infinite reconnection on desktop web browsers, when accessing to an iPhone it works

SpaghettiC0des, 2023-10-16:
### What happened

This is similar to https://github.com/Arksine/moonraker/issues/454, but it works on an iPhone, but when you access it on a MacOS, regardless of the browser used, it will result in an infinite loop of handshake and disconnection.

You can see in the logs the iPhone is working correctly while the MacOS web is dropping the connections without any reason.

### What did you expect to happen

It should work on port 80 on desktop web browsers. 

### How to reproduce

1. Follow RatRig's installation instructions https://os.ratrig.com/docs/introduction/
1. After the setup, open http://ratos.local or http://<IP_ADDRESS>.
1. Open the networks tab, and see the WebSocket infinite reconnection request
1. Only port 80 does this, but when you visit port 7125, or ratos.local/configure it works perfectly.

### Additional information

See ticket https://github.com/Arksine/moonraker/issues/726

And [comment from the maintainer (Arksine) ](https://github.com/Arksine/moonraker/issues/726#issuecomment-1763461921)

miklschmidt, 2023-10-17:
if `/configure` works, then it points to a mainsail issue since the configurator connects to moonraker as well. However, i haven't heard about this particular problem before, it could be a local issue, i don't know. Unless you have some indication that it's a RatOS specific problem, there's not much i can do. Have you tried a plain MainsailOS installation?

SpaghettiC0des, 2023-10-17:
@miklschmidt , not sure if you saw it, but the Moonraker author asked me to submit an issue here because RatOS might be using an NGINX reverse proxy. 
 
I added a comment there https://github.com/Arksine/moonraker/issues/454, where the hosted mainsail works. 

I'll try plain MainsailOS tomorrow.

miklschmidt, 2023-10-17:
Same reverse proxy as MainsailOS, also it works for everyone else. Seems to be MacOS specific?

SpaghettiC0des, 2023-10-17:
> Same reverse proxy setup as MainsailOS. Also it works for everyone else. Seems to be MacOS specific?

Maybe? I'll try an older intel MacBook tomorrow, both of my M1 Macs are showing the same behavior.