Rat-OS/RatOS #109: LDO Orbiter V2.0 - Configs deviate from OEM suggestions

swilkens, 2023-11-09:
### Is your feature request related to a problem? Please describe

The TMC drivers settings and run current for the LDO Orbiter 2.0 with the LDO-36STH20-1004AHG motor deviate from LDO's specification. This has been done intentionally, but that isn't clear from the configuration files or the documentation.

LDO Suggestions:
[image]

RatOS settings:
[image]
[image]

### Describe the solution you'd like

Add a note to the configuration file for the Orbiter v2.0 that clarifies that these changes are by design.

### Additional information

As discussed on Discord with @miklschmidt

HelgeKeck, 2023-11-09:
run current 0.85 is waya too high and absolutely not needed for a v-core
mikl did this for reasons

swilkens, 2023-11-09:
Like I said in the issue report, this just serves as a reminder for mikl to make a note in the config file or something :)

miklschmidt, 2023-11-11:
Yeah basically my explanation here: https://discord.com/channels/582187371529764864/1092127173223125163/1171965371004231730

Should make it into comments or docs somewhere.

miklschmidt, 2024-02-10:
As 2.1 handles current and presets quite differently, this should be more self explanatory. The configurator will warn about overcurrenting if current is set beyond the rated current of the stepper. It also make it much easier to change, should the user want to try overcurrenting as per OEM suggestions anyway.

miklschmidt, 2024-06-14:
Closed following 2.1.0-RC1

closed 2024-06-14 by miklschmidt