Rat-OS/RatOS #113: Prime blob or prime line can exceed maximum extrusion (esp. with small nozzles)

david-antos, 2023-12-25:
### What happened

Switching to 0.25mm nozzle, I got
```
Move exceeds maximum extrusion (0.289mm^2 vs 0.250mm^2)
See the 'max_extrude_cross_section' config option for details
```
during prime blob. I tried prime line as well, which pushes even more filament, resulting in the same error.
I have correct nozzle diameter in printer.cfg (as well as in my slicer setting, which is probably not relevant as priming is controlled solely by printer configuration). In addition to that, when priming is set to False, there are no issues with the nozzle (suggesting my setup is likely fine otherwise).

What is even worse, the error is displayed, printing stops, and both bed and extruder heaters stay on.

### What did you expect to happen

Should the maximum extrusion be exceeded, RatOS should at least end up in shutdown (not keeping the extruder in temperature).

Priming the nozzle should be possible regardless of nozzle size.

### How to reproduce

Start the print under conditions described in the "what happened" section. The process is deterministic. See also additional info for my primary analysis.

### Additional information

Basic analysis: the speeds of priming line and blob extrusion are derived as 60 percent of macro_travel_speed (I have the default configured, i.e. 300). This is a bit unfortunate as the variable is primarily used for actual non-extrusion movement in macros, priming is probably the only case when maximum extrusion can be exceeded. I'd suggest deriving the speed of priming as, e.g. the minimum of {macro_travel_speed, Ve}, where Ve = 90% of "what the maximum extrusion allows". This way, the extrusion would be controlled by the parameters of the nozzle (with a safety margin), and limited by macro_travel_speed.

miklschmidt, 2024-01-02:
Prime blob needs extrusion calculation based on nozzle size. It should however still work fine with 0.25mm nozzles if you leave the nozzle_diameter at 0.4, i'd use that as a workaround for now.

The only thing nozzle_diameter is used for in klipper is to calculate the current threshold to throw that error.

> What is even worse, the error is displayed, printing stops, and both bed and extruder heaters stay on.

That's how klipper behaves on non fatal errors. You'd have to raise an issue over there.

miklschmidt, 2024-02-10:
in 2.1 (still unreleased), primeblob extrusion is now scaled by nozzle diameter (which is set in the configurator).

miklschmidt, 2024-06-14:
Closed following 2.1.0-RC1

closed 2024-06-14 by miklschmidt