Rat-OS/RatOS #114: update error with klipper(v0.11.0-219 > v0.12.0-40) and moonraker  (v0.8.0-48 > v0.8.0-231)

DarkRabbit19, 2023-12-29:
### What happened

After the data has been fetched for klipper and/or moonraker. The update fails with the error:  "reference broken".
(this happend on a clean installation).
[Update_log.txt](https://github.com/Rat-OS/RatOS/files/13792080/Update_log.txt)

### What did you expect to happen

That after the update, klipper and moonraker would be the latest version.

### How to reproduce

Clean install on the "Raspberry Pi 3 B" and then in the "MACHINE" tab "update al components".

DarkRabbit19, 2024-01-02:
problem was fixed with a other re instal of the RATOS software
there is now mentioning of  "untracked source files" but this should not give any problems.
[image]

closed 2024-01-02 by DarkRabbit19

miklschmidt, 2024-01-02:
For future reference, that error indicates a dying sd card. I'd swap to a new one from a reliable brand (can recommend SanDisk Extreme).