Rat-OS/RatOS #120: Octopus max support

mkljax, 2024-03-06:
### What happened

not seeing Octopus Max for a control board in printer setup with the armbian CB1 Ratos V2.0.2

### What did you expect to happen

expected to see Octopus Max EZ as a printer setup option because it is listed on the url https://os.ratrig.com/docs/boards/list

### How to reproduce

login into " http://ratos.local/configure?step=1" and do not see Octopus Max EZ as a printer setup choice.

### Additional information

Tried to do a manual klipper firmware compilation but could not find octopus max ez in the "boards" directory.

keyquesttech, 2024-03-11:
Did you manage to fix it?

mkljax, 2024-03-12:
No. It may not be implemented because the CB1 is experimental.

keyquesttech, 2024-03-12:
I managed to do it. Just had to log in to the mainsail GUI and update everything

mkljax, 2024-03-12:
Did you use the 32bit or 64bit version of armbian?

mkljax, 2024-03-12:
Sorry thought there was a 32bit armbian, its only the Pi that has both.

mkljax, 2024-03-12:
Thx, that got it to populate with the board.

closed 2024-03-12 by mkljax