Rat-OS/RatOS #126: Support BTT CB2

vzagranichnyy, 2024-05-31:
### Is your feature request related to a problem? Please describe

Is support BTT CB2 expected?

### Describe the solution you'd like

support BTT CB2

lstandifer, 2024-11-03:
CB2 news ?

https://github.com/Rat-OS/RatOS/pull/162

3dprintpt, 2024-11-04:
> CB2 news ?
> 
> #162

yes... Working on it with Mikkel (he did almost everything :P )
Will be testing tomorrow

trudeau87, 2024-11-13:
How is the testing going?

3dprintpt, 2024-11-13:
Not as fast as we wanted... Other things in the way, sorry.
There is someone already using the "handmade" image I built last week.
If you want to give it a try let me know, please.

trudeau87, 2024-11-13:
Id love to!  I have been trying to install this for weeks. ;-)

3dprintpt, 2024-11-13:
https://mega.nz/file/XQkDSBxb#JOtfpw2w4Iy4dUpCWg3pQEzX6Av0vkj9FlBWCVhLsXw

Here you can download the image.

After flashing the sdcard, you need to edit the system.cfg file to setup your internet, like on normal CB2 system. Not the best practice but should work.

lstandifer, 2024-11-14:
works well seems to run better than the stock armbian image from btt

sylanc3, 2025-05-14:
Hello together, since I think my CB1 is causing often MCU timeout issues, I want to upgrade to a CB2. Is there an actual / recent image for RatOS 2.x for the CB2 available?
Thanks for your support.
BR Flow

bauderline, 2025-06-17:
I used the above mega link to download the image but if I try to mount it or burn it to SD it complains it is corrupted. Is there another version of this image anywhere ? Thanks!

bauderline, 2025-06-17:
That is exactly what I did do, extract the file and then use balena etcher to write to the sd card. it comes back as corrupt each time after the verify. When I try to mount the disc image it also says it is corrupt. Have tried different computers and different SD cards, one 32gb the other 64gb.

sylanc3, 2025-06-17:
> That is exactly what I did do, extract the file and then use balena etcher to write to the sd card. it comes back as corrupt each time after the verify. When I try to mount the disc image it also says it is corrupt. Have tried different computers and different SD cards, one 32gb the other 64gb.

So as it worked fine for me, we can be sure that the file on Mega seems to be fine.
maybe redownload it again, decompress it and maybe use dd-comand on linux or unix system to copy to your SD-Card