Rat-OS/RatOS #127: PSA: RatOS already has Beacon installed and will flash it automatically, don't install it again.

miklschmidt, 2024-06-01:
With the release of Beacon TrueZero, a lot of people are running the beacon install scripts on their setups. This is not necessary and breaks the existing beacon installation. All you need to do is make sure your beacon is plugged in, then update RatOS and Beacon through the mainsail machine tab, and RatOS will flash your beacon automatically.

WO1984PR, 2024-11-14:
So, when we see the message "beacon v2.0.0-16 > v2.0.0-17" "Update" Ignore it? My current version is "mcu beacon(beacon)
Version: Beacon 2.1.0"  just from doing all of the other updates on first boot.

miklschmidt, 2024-11-15:
You should still update it through the update manager :)

WO1984PR, 2024-11-15:
I am just trying to get my install to work. But, I did the upgrade from this page in the machine.
[image]

closed 2024-12-19 by miklschmidt