Rat-OS/RatOS #128: RatOS 2.1-RC1

MintyTrebor, 2024-06-01:
### What happened

Used configurator to setup Vcore 3.1 with SKR Pro 1.2, RatOS.cfg sets some wrong pins for the board which leads to Klipper not starting. Also in the wizard, setting- Electronics -> controller fan to **_No Fan_** still leads to pins being assigned to fan_controller_board_pin alias - if the alias is removed klipper errors due to missing alias - so I am unable to use that pin for a different type of fan config:

Relevant sections from RatOS.cfg:
```
# WARNING. THIS FILE IS GENERATED BY THE RATOS CONFIGURATOR.
# CHANGES YOU MAKE HERE WILL BE OVERWRITTEN. KEEP YOUR CHANGES IN PRINTER.CFG.
# Config generated for Rat Rig V-Core 3 400
# Documentation: https://os.ratrig.com

#############################################################################################################
### CONTROLBOARD & TOOLBOARD
#############################################################################################################
[include RatOS/boards/rpi/config.cfg]

[board_pins btt-skr-pro-12]
aliases:
	#----------------------------------------------- X motor pins -----------------------------------------------
	# Assigned to slot: "X"
	#------------------------------------------------------------------------------------------------------------
	x_step_pin=PE9,
	x_dir_pin=PF1,
	x_enable_pin=PF2,
	x_uart_pin=PC13,
	x_diag_pin=PB10,
	x_endstop_pin=PB10,
	#----------------------------------------------- Y motor pins -----------------------------------------------
	# Assigned to slot: "Y"
	#------------------------------------------------------------------------------------------------------------
	y_step_pin=PE11,
	y_dir_pin=PE8,
	y_enable_pin=PD7,
	y_uart_pin=PD11,
	y_diag_pin=PE12,
	y_endstop_pin=PE12,
	#----------------------------------------------- Z motor pins -----------------------------------------------
	# Assigned to slot: "E0"
	#------------------------------------------------------------------------------------------------------------
	z0_step_pin=PE14,
	z0_dir_pin=PA0,
	z0_enable_pin=PC3,
	z0_uart_pin=PC7,
	z0_diag_pin=PE15,
	#---------------------------------------------- Z1 motor pins -----------------------------------------------
	# Assigned to slot: "E1"
	#------------------------------------------------------------------------------------------------------------
	z1_step_pin=PD15,
	z1_dir_pin=PE7,
	z1_enable_pin=PA3,
	z1_uart_pin=PF2,
	z1_diag_pin=PE10,
	#---------------------------------------------- Z2 motor pins -----------------------------------------------
	# Assigned to slot: "E2"
	#------------------------------------------------------------------------------------------------------------
	z2_step_pin=PD13,
	z2_dir_pin=PG9,
	z2_enable_pin=PF0,
	z2_uart_pin=PE4,
	z2_diag_pin=PG5,
	#------------------------------------------- EXTRUDER motor pins --------------------------------------------
	e_step_pin=PE13,
	e_dir_pin=PC2,
	e_enable_pin=PC0,
	e_uart_pin=PE1,
	#----------------------------------------------- GENERAL PINS -----------------------------------------------
	e_heater_pin=PB1,
	e_sensor_pin=PF4,
	stepper_spi_mosi_pin=null,
	stepper_spi_miso_pin=null,
[... 328 lines cut ...]
stealthchop_threshold: 0
interpolate: False
uart_pin: PD6
run_current: 1.6
sense_resistor: 0.11
driver_TBL: 2
driver_TOFF: 3
driver_HEND: 0
driver_HSTRT: 6

[stepper_z2]
step_pin: PD13
dir_pin: PG9
enable_pin: !PF0
microsteps: 64
full_steps_per_rotation: 200
rotation_distance: 4
```

### How to reproduce

Configer printer vcore 3.1 400 using configurator wizard
FYI: I also use a beacon and ebb36 1.2 toolboard (just in case that is somehow involved in triggering the configurator issue)

I also checked the _config.cfg_ file for the _btt-skr-pro-12_ and the pins seem to be defined correctly.

miklschmidt, 2024-06-01:
> Used configurator to setup Vcore 3.1 with SKR Pro 1.2, RatOS.cfg sets some wrong pins for the board which leads to Klipper not starting

Please specify exactly what is assigned incorrectly and what you expected.

>  setting- Electronics -> controller fan to No Fan still leads to pins being assigned to fan_controller_board_pin alias and it being linked to the mcu temp - if the alias is removed klipper errors due to missing alias.

I should probably explain a few things here, as it's not very obvious.

1) The controller fan has nothing to do with mcu temperature.
2) RatOS 2.1 is supposed to be backwards compatible with 2.0 `printer.cfg`s
3) The V-Core 3 configuration files existed prior to 2.1
4) The V-Core 3 configuration files includes a `[controller_fan]` definition, which cannot be removed without breaking backwards compatibility with 2.0 `printer.cfg`s
5) The board pin aliases written to `RatOS.cfg` is exclusively for backwards compatibility and to abstract the user overridable convenience settings written to `printer.cfg` (so they wouldn't trigger a `printer.cfg` change when swapping boards - there are still other issues with this, and these "convenience" settings will be removed entirely in the future, replaced by a dedicated configurator interface, but i digress).

Given these conditions, the V-Core 3 (and other 2.0 supported printers) does not currently support not having a controller fan defined. If you pick "No fan", it'll write the default pin to the pin alias (because backwards compatibility).

I have no good way to detect this, you could argue the bug is that "No fan" is shown as an option for printer configs that actually don't support a "No fan" option, however this doesn't actually cause any issues unless the user decides to use the default controller fan port for something else (disregarding the wiring diagram). However the configurator now knows you're not using a controller fan so we can act on that information for new features and functionality in the future, such as progressively cleaning up the old configuration files, as we move towards **fully** generated configs.

So, to sum it all up, it's a feature, not a bug, just ignore it :)

MintyTrebor, 2024-06-02:
Below is a list of the pins changed with the original RatOS.cfg value shown in brackets. The values I have changed are according to the SKR Pro 1.2 pin out diagram, and the pins in the config.cfg file for the board.

```
[board_pins btt-skr-pro-12]
aliases:
	#----------------------------------------------- Y motor pins -----------------------------------------------
	# Assigned to slot: "Y"
	#------------------------------------------------------------------------------------------------------------
	y_uart_pin=PE3 (PD11),
	y_endstop_pin=PE2 (PE2),
	#----------------------------------------------- Z motor pins -----------------------------------------------
	# Assigned to slot: "E0"
	#------------------------------------------------------------------------------------------------------------
	z0_uart_pin=PD4 (PC7),
	#---------------------------------------------- Z1 motor pins -----------------------------------------------
	# Assigned to slot: "E1"
	#------------------------------------------------------------------------------------------------------------
	z1_uart_pin=PD1 (PF2),
	#---------------------------------------------- Z2 motor pins -----------------------------------------------
	# Assigned to slot: "E2"
	#------------------------------------------------------------------------------------------------------------
	z2_uart_pin=PD6 (PE4),
	
#############################################################################################################
### STEPPER MOTORS, DRIVERS & SPEED LIMITS
#############################################################################################################
#---------------------------------------------------- Y -----------------------------------------------------
# The B motor in the CoreXY system, located at the rear right of the printer
# Connected to Y on SKR Pro v1.2
#------------------------------------------------------------------------------------------------------------
[tmc2209 stepper_y]
uart_pin: PE3 (PD11)

#---------------------------------------------------- Z -----------------------------------------------------
# The left Z motor for the kinematic bed
# Connected to E0 on SKR Pro v1.2
#------------------------------------------------------------------------------------------------------------
[tmc2209 stepper_z]
uart_pin: PD4 (PC7)

#---------------------------------------------------- Z1 ----------------------------------------------------
# The rear Z motor for the kinematic bed
# Connected to E1 on SKR Pro v1.2
#------------------------------------------------------------------------------------------------------------
[tmc2209 stepper_z1]
uart_pin: PD1 (PF2)

#---------------------------------------------------- Z2 ----------------------------------------------------
# The right Z motor for the kinematic bed
# Connected to E2 on SKR Pro v1.2
#------------------------------------------------------------------------------------------------------------
[tmc2209 stepper_z2]
uart_pin: PD6 (PE4)
```

I understand the constraint with the fan pin. Is there a recommeded approach to re-assigning a fan pin? - I need to use one for a ptc chamber heater fan control. Any fan pin will do as I do not use any of the SKR Pro's fan pins for my other fans...

MintyTrebor, 2024-06-05:
Figured out the error is in the board-definition.json, so I updated it and created a pull request
[267](https://github.com/Rat-OS/RatOS-configuration/pull/267)

miklschmidt, 2024-06-06:
> I understand the constraint with the fan pin. Is there a recommeded approach to re-assigning a fan pin? - I need to use one for a ptc chamber heater fan control. Any of the three fan pins will do as I do not use any of the SKR Pro's fan pins for my other fans...

You can use a sacrificial pin (an unused port you don't use), for example on the rpi, to free up the pin used for the controller fan:
```properties
[controller_fan controller_fan]
pin: rpi:gpio4
```

closed 2024-06-06 by MintyTrebor