Rat-OS/RatOS #130: Rear right steppers do not move on hybrid config while X axis movement

macke4ever, 2024-06-02:
### What happened

V-core 4 500mm hybrid setup, when configured to work as hybrid some stepper motors do not move. Y axis works fine, while X axis rotates only left side steppers. Thus toolhead moves diagonaly only

### What did you expect to happen

When toolhead moved in X axis all 4 rear steppers should rotate

### How to reproduce

Rat OS 2.1, vcore 4 500mm, hybrid configuration. Move toolhead left/right.

Myles5588, 2024-06-09:
Wiring diagram appears to not match up with what is in the printer.cfg file.  Match your stepper/board wiring to what is in the printer.cfg file.  I found I had to switch motor1 and motor4 on my octopus board.

miklschmidt, 2024-06-13:
I forgot to reply on this. The problem is indeed the official wiring diagram is incorrect. You can remap your steppers using "advanced stepper configuration" in the configurator, or move the cables around. For hybrid the order (from left to right) is X(A), Y, E,X1(B),Y1.

closed 2024-06-13 by miklschmidt