Rat-OS/RatOS #131: Installing RatOS 2.1 on Prusa MK3S: no way to flash Einsy board

david-antos, 2024-06-09:
### What happened

I'm trying to install RatOS 2.1 pre-release onto Prusa MK3S, the Einsy board bootloader is already modified to handle Klipper and v0.12.0-207 is flashed there (I happily run it with RatOS 2.0).
The RatOS configuration wizard allows me to select the printer, to select Einsy board in the following step. In the "Control board flashing" step, see screenshot.
[image]

OK, SD card is the only way to proceed, so I expect I can cheat. I compiled the firmware manually and flashed it. Then I turned the printer off and on again and tried "Check board status". It does nothing and it is not possible to proceed (as "Next" button is inactive).

### What did you expect to happen

Ideally, the wizard should be able to flash the board directly (automatic updates work on RatOS2.0). Alternatively, there may be an option for strange boards like Einsy "The board is already flashed, I know what I'm doing, let me finish the wizard and create the configuration".

### How to reproduce

Running the config wizard in RatOS 2.1.

miklschmidt, 2024-06-13:
I suspect something else is wrong here.

> Ideally, the wizard should be able to flash the board directly (automatic updates work on RatOS2.0). Alternatively, there may be an option for strange boards like Einsy "The board is already flashed, I know what I'm doing, let me finish the wizard and create the configuration".

If the configurator doesn't detect it, klipper will throw errors. There's no way around it. The fun part here is that the board isn't detected at all, not for automatic flashing (ie. it had firmware with the prusa-einsy chip ID in there) and not as an unidentified device. It looks like it's not connected at all (or the board definition has an error, which would be strange as i don't believe it has changed). Can you run `lsusb` and check if it connects at all?

david-antos, 2024-06-13:
Here we go:
```
pi@prupi:~ $ sudo lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 046d:0990 Logitech, Inc. QuickCam Pro 9000
Bus 001 Device 003: ID 2c99:0002 Prusa Original Prusa i3 MK3
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
and also
```
pi@prupi:~ $ ls -la /dev/serial/by-id/usb-Prusa_Research__prusa3d.com__Original_Prusa_i3_MK3_CZPX0119X004XK02510-if00 
lrwxrwxrwx 1 root root 13 Jun  9 13:05 /dev/serial/by-id/usb-Prusa_Research__prusa3d.com__Original_Prusa_i3_MK3_CZPX0119X004XK02510-if00 -> ../../ttyACM0
pi@prupi:~ $ ls -l /dev/ttyACM0 
crw-rw---- 1 root dialout 166, 0 Jun  9 13:05 /dev/ttyACM0
```
and even
```
cd ~/klipper
cp -f ~/printer_data/config/RatOS/boards/prusa-einsy/firmware.config ~/klipper/.config
make olddefconfig
make clean
make
sudo service klipper stop
make flash FLASH_DEVICE=/dev/serial/by-id/usb-Prusa_Research__prusa3d.com__Original_Prusa_i3_MK3_CZPX0119X004XK02510-if00
```
succeeds hapilly (with avrdude saying verified/fuses OK/done). That seems pretty solid to me :).

miklschmidt, 2024-06-13:
Aw dangit. I forgot there's no chipid to set in atmega firmware.. @2bbe1c8 should fix it, may require a reboot after updating to detect correctly though.

david-antos, 2024-06-14:
Great, I can confirm it's working now, it allows flashing the firmware directly from the wizard. Thank you!

closed 2024-06-14 by david-antos