Rat-OS/RatOS #132: Support for OrangePi cm4 in Ratos 2.1

cross81, 2024-06-09:
### Is your feature request related to a problem? Please describe

I currently run my RatRig with an OrangePi cm4, and it would love to be able to use RatOS again.

### Describe the solution you'd like

Add support for the OrangePi cm4 debian http://www.orangepi.org/html/hardWare/computerAndMicrocontrollers/service-and-support/Orange-Pi-CM4-1.html to RatOS 2.1

miklschmidt, 2024-06-13:
Oh they have a raspberry pi image. In that case it might be straight forward, i've started a build here: https://github.com/Rat-OS/RatOS/actions/runs/9505229132/job/26199647562

miklschmidt, 2024-06-13:
It's missing a standard boot/config.txt .. i'll add the [all] sections from the original image and see what happens.

miklschmidt, 2024-06-13:
Trying again https://github.com/Rat-OS/RatOS/actions/runs/9505585306

cross81, 2024-06-13:
> It's missing a standard boot/config.txt .. i'll add the [all] sections from the original image and see what happens.

they renamed it  /boot/orangepiEnv.txt
took me a minute to figure that one up when i was setting up a BTT HDMI7

miklschmidt, 2024-06-13:
@cross81 different file (also wouldn't do anything on raspberry pi os), commandline.txt is missing too.

miklschmidt, 2024-06-13:
Here we go again https://github.com/Rat-OS/RatOS/actions/runs/9508255431

miklschmidt, 2024-06-14:
@cross81 aight... it finished.. Give it a shot! https://github.com/Rat-OS/RatOS/actions/runs/9508255431/artifacts/1600396186

cross81, 2024-06-14:
Awesome! Thank you! I'll try it out.

cross81, 2024-06-14:
It works! took me a bit to get a can network set up until I realized that /etc/network/interfaces was missing the line source /etc/network/interfaces.d/*
once I fixed that, everything worked great!

Thank You again

miklschmidt, 2024-06-14:
I should probably add the interfaces.d/* inclusion, it's been missing for a while (not sure why it's not default in the base image), and since we prefer USB it doesn't impact RatOS otherwise, so i keep forgetting :D

Good to hear though, will include it by default from here on out, at least until we switch to bookworm (unless they follow suit).

closed 2024-06-14 by miklschmidt

cross81, 2024-06-15:
They already have bookworm images available. :)