Rat-OS/RatOS #135: ratos.cfg needs to be read only for user space or allow for user consent for a writable file

smwoodward, 2024-06-21:
### What happened

The writeable file ratos.cfg was over written without user consent

### What did you expect to happen

ratos.cfg should be read only for user space

### How to reproduce

This is self explanatory 

### Additional information

ratos.cfg is a writeable file in user space.  If this file is to be automatically updated it needs to be a read only file for the user space.  The updating mechanisms and scripts that run on the system should be either run as a different user or an elevated user to allow editing the files for updates and to keep the file as read only for user space.

miklschmidt, 2024-06-21:
First 2 lines in RatOS.cfg:
```
# WARNING. THIS FILE IS GENERATED BY THE RATOS CONFIGURATOR.
# CHANGES YOU MAKE HERE WILL BE OVERWRITTEN. KEEP YOUR CHANGES IN PRINTER.CFG.
```

This is by design, when something changes in RatOS, RatOS.cfg gets regenerated. If you screw over RatOS.cfg, it gets regenerated - you get the idea. The configurator makes sure RatOS.cfg does what it's supposed to do.

User space = moonraker, klipper, ratos-configurator etc. "But you should really setup users for each of these services". Yes, i should, is it something that makes any sense to prioritise for me when there are real usability issues to solve? Not really. It brings no real value to the user. Moonraker is the endpoint for file manipulation through mainsail, it automatically refuses to edit files within the directories of clients registered with it's update manager. RatOS.cfg is just a generated flat file, i chose to place it next to printer.cfg so the user had easy access to read it, in case there was something they wanted to understand, and to copy paste from it, in case there was something they wanted to modify.

A "proper" permission hierarchy makes it harder to accomplish an automatic installation procedure (so you can install RatOS on any platform) that others have requested. However, if someone wants to do all the grunt work and submit a well written PR to address it i'll gladly accept it. If not, it'll happen when it's blocking further progress towards the main goal of RatOS. It does not have priority right now.

So for now, the advice is: Do as instructed, don't modify RatOS.cfg and expect the changes to persist.

That said, making the file read only is not a bad idea, and i would've done so if it was a quick fix, unfortunately moonraker currently doesn't have a handle to make that happen at this point, at least not one that doesn't involve moving the file further out of reach of the user. When such a handle is implemented, i'll mark it read-only.

Closing this for now, as it's working as intended.

closed 2024-06-21 by miklschmidt

smwoodward, 2024-06-21:
There are plenty of other moonraker and klipper tools that update through moonraker that is able to update read only files in user space automatically if they are edited at the repo.  This may work as designed, but the design is bad for users.  This is why it should be a read only file for the user.

smwoodward, 2024-06-21:
@miklschmidt is RatOS-Configurator able to run scripts on the OS?

miklschmidt, 2024-06-21:
@smwoodward I don't think you read what i just wrote. Moonraker automatically disallows writing to files residing in repositories registered via the update_manager, it's not something those modules do. RatOS.cfg does not live in a repository.

> This may work as designed, but the design is bad for users
There's been 3 instances of people who didn't read those first 2 lines, and were confused why it was overwritten, and that is **without** the docs for 2.1 being published. I think you're making a very big deal out of a relatively minor issue.

miklschmidt, 2024-06-21:
> @miklschmidt is RatOS-Configurator able to run scripts on the OS?

Yes, and i could micro manage the permissions by writing a bash script that needs to run as root and whitelist it for passwordless sudo, but that's a huge hack that has potentially problematic side-effects and you know it.

miklschmidt, 2024-06-21:
There's a quick fix... I could move RatOS.cfg into RatOS/ and add it to .gitignore in RatOS-configuration. Still a bit of a hack, but doesn't have any nasty side effects. Well except that it would break all the printer.cfgs already running, would need to still generate RatOS.cfg and have it include RatOS/RatOS.cfg .. I'll think about it.

smwoodward, 2024-06-22:
> > @miklschmidt is RatOS-Configurator able to run scripts on the OS?
> 
> Yes, and i could micro manage the permissions by writing a bash script that needs to run as root and whitelist it for passwordless sudo, then run it before and after writing to RatOS.cfg, but that's a huge hack that has potentially problematic side-effects and you know it.

You don't need to be defensive like this bud.  I'm trying to help you fix issues that have been seen down the line.  I was hoping that since ratos.cfg wasn't a read only file that would make issues like different communications options between the SBC and the MCU and still getting RatOS updates a non issue, however I did learn that Configurator overwrites the file on it's own.  Yes there is the comment that states that it will be overwritten in ratos.cfg but with the diff comparison in Configurator it appears that ratos.cfg wouldn't be over written automatically, only when I choose that in Configurator and click submit.  Standard users will not understand fully why a non read only file gets over written without them doing something.  Like printer.cfg isn't over written without a user being able to see it and their action caused the file to be change.  I think this would be a better thing to do for ratos.cfg and maybe create a moonraker alert if there isn't an approved diff in ratos.cfg before the file is changed without user action, but at least marking it read only would prevent users from not knowing what happened when one day their printer worked and the next day it didnt.

The design logic of RatOS with ratos.cfg is a sound one in users being able to get updates for incorrect pin assignments or other issues that may come up, and I know that a user could make a copy of ratos.cfg into user-ratos.cfg and change the include in printer.cfg unfortunately would exclude users from getting all of the updates that RatOS might add or change in updates.  I think eventually adding a file diff tracking system for ratos.cfg and printer.cfg would add ""support"" for different configuration changes without fully supporting anything outside of USB devices.  That way if there is an update to the pin configuration in ratos.cfg, but a user changed the connection to the mcu for canbus, uart, or even if klipper adds some sort of communication method in the future, that line in ratos.cfg could be logged and when ratos.cfg is updated  to fix the pin configuration, the users edits would be preserved.  I do understand that this would be significantly more work which is why I think just making ratos.cfg read only would be the best bet to do now.

I do understand the issue in 2.1 instances already installed that they would need to have RatOS/RatOS.cfg added to printer.cfg.  You could possibly use a script with **sed** to search for and change the line, or it's still a release candidate and put a note about it in the final release

smwoodward, 2024-06-22:
> @smwoodward I don't think you read what i just wrote. Moonraker automatically disallows writing to files residing in repositories registered via the update_manager, it's not something those modules [...]

I think that the number of people who didn't read those first 2 lines will only increase more once it's out of RC and the Doc's will likely not help.  

Unless it's in Bold and large red print on ever page.

miklschmidt, 2024-06-22:
> You don't need to be defensive like this bud.

You don't need to be so argumentative. We don't disagree about the issue, we disagree about the importance relative to the other things on my todo list.

> Yes there is the comment that states that it will be overwritten in ratos.cfg but with the diff comparison in Configurator it appears that ratos.cfg wouldn't be over written automatically, only when I choose that in Configurator and click submit.

You're mixing RatOS.cfg and printer.cfg. RatOS is **always** overwritten when changing the hardware configuration unless you explicitly ignore the changes - you're not asked to make a decision. It doesn't care about user modifications. Only files meant for user modifications such as printer.cfg explicitly requires confirmation before they're overwritten.

> I think this would be a better thing to do for ratos.cfg and maybe create a moonraker alert if there isn't an approved diff in ratos.cfg before the file is changed without user action, but at least [...]

I don't know why you're still arguing. I don't disagree, i'm just not going to do it at any cost, it's a minor issue and everything is a cost-benefit analysis when time is limited. And yet here i am still going in circles with you again. There are much bigger fish to fry, when there's a good option for having moonraker treat the file as read only, it will be done. Until then, it's going to be mentioned in the docs and in the file itself. End of story.

> I think eventually adding a file diff tracking system for ratos.cfg and printer.cfg would add ""support"" for different configuration changes without fully supporting anything outside of USB devices.

This is an absolute nightmare to maintain and will never work in all cases. One week in a team where multiple people are changing the same file, and you would know why. There's no reliable way to merge all conflicts automatically, it's a problem that has never been fully solved, even through decades of version control developments.

RatOS is built around USB, users who still want to go down other paths will have to find their own workarounds. I'm not going to argue with you over this again. That discussion is dead.

This will be my last entry here. It will be fixed when i decide it makes sense to fix it.

smwoodward, 2024-06-22:
> > You don't need to be defensive like this bud.
> 
> You don't need to be so argumentative. We don't disagree about the issue, we disagree about the importance relative to the other things on my todo list.
> 

I'm not arguing with you on your todo list, but these issues are very likely to be a lot more relevant than you make them out to be.

> You're mixing RatOS.cfg and printer.cfg. RatOS is **always** overwritten when changing the hardware configuration unless you explicitly ignore the changes - you're not asked to make a decision. It [...]

I am not mixing them up.  I am pointing out an issue and explaining that it wouldn't be a bad idea to eventually add a way to allow for some hardware configurations that RatOS-Configurator does not support that could only be changed in ratos.cfg.  However, if this file is going to be editable by a user in mainsail, as you already said, 2 people have already reached out to you about this issue.  So either allow 2 files meant for user modifications, or make the files not meant for user modifications to be read only.  If it's not meant for user modifications because of your design, then the file should not be user editable.

> I don't know why you're still arguing. I don't disagree, i'm just not going to do it at any cost, it's a minor issue and everything is a cost-benefit analysis when time is limited. And yet here i am [...]

Again, this is likely to be a much bigger issue than you think, AND it would also add to the ability for the user to modify AND maintain the updated benefits of RatOS.  Until then, no matter how much you mention it in the docs, you will keep answering these issues.  If you make it read only, HOWEVER  you want or feel like doing it, that will STOP these issues.  Period, End of Story.  Otherwise, you'll keep getting them and having to explain it.

> This is an absolute nightmare to maintain and will never work in all cases. One week in a team where multiple people are changing the same file, and you would know why. There's no reliable way to [...]

I deal with system related issues for 20+ developers.  I understand the merge conflicts especially in several git based systems.  Tracking the edits in 1 file, used by 1 user, for 1 purpose that is edited infrequently is nowhere like dealing with 20+ developers even for an hour.

> RatOS is built around USB, users who still want to go down other paths will have to find their own workarounds. I'm not going to argue with you over this again. That discussion is dead.
> 
> This will be my last entry here. It will be fixed when i decide it makes sense to fix it.

And this is what I'm saying about you being so defensive.

Smwoodwardagain, 2024-06-22:
@miklschmidt in an era of the internet, where creating accounts is easy and free, blocking people doesn't stop the potential of the conversation from continuing. It only gives the administrator a little better feeling about himself and that he thinks he has some power. You know I'm correct in what needs to be done and you also know I'm correct in what should be done in the future. However you want to fight against anything but USB and will use your software to push that agenda. 

So go ahead and block this too if it makes you feel better, I will still be around and if I want to comment, you can't stop it.

Smwoodwardagain, 2024-06-22:
@miklschmidt now that's funny. You marked my comment "is RatOS-Configurator able to run scripts on the OS?" As abuse. You really have tender feelings about things that go against your line of thinking?

smwoodwardx2, 2024-06-22:
@miklschmidt Spam is not really a good marker for those messages.  Are you franticly clicking buttons now to stop getting notification?  Now the Second messages would have been a good marker as abuse, but they weren't a good fit for the first ones, unless getting your feelings hurt by someone not kissing your boot and calling you out on your bias and misinformation is what you consider abuse.  

I gave you the opportunity to try and hash this out privately, but instead you didn't want me to DM you, and you even ridiculed me.  So here we are.

smwoodwardx2, 2024-06-22:
@miklschmidt Now, if you would actually like for me to stop this one way or another, lets talk.  Privately and peacefully to come to a compromise, unless your bias will not allow you to even make a compromise.

smwoodwardx2, 2024-06-22:
@miklschmidt  however I k is you won’t attempt to even have a conversation because your position and “reasons” for things are not defendable.

smwoodwardx2, 2024-06-22:
@miklschmidt I will give you another instance where you and I butted heads and I essentially had to make you listen when when I found a bug with z_thermal_adjust.  The bug ended up being in klipper and not RatOS, but you argued with me for DAYS that it was something in my configuration, and you had all of the people backing you up blah blah blah that it was my config.  https://klipper.discourse.group/t/bug-bed-mesh-and-z-thermal-adjust-applies-z-changes-even-if-z-isnt-homed/7415?page=2. 

You don't listen to your users.  And as a representative of RatRig (and you are part of RatRig now, like it or not, even if you are not employed by them) You can't take the little niche community project mentality anymore.  You are going to be a tarnish on RatRig.

smwoodward-beep, 2024-06-23:
@miklschmidt that's better on a more appropriate marker, which shows you are thinking about what you're reading.  Further adding the enhancement tag does indicate that you know I'm right, even though it's more telling that you don't consider over writing a user writable file a bug, but a feature.