Rat-OS/RatOS #138: Built in filament unload causing clogs

smwoodward-beep, 2024-06-25:
### What happened

The tip forming dipping done prior to fully ejecting the filament with an Orbiter 2 and Phaetus UHF Rapido pulls up molten filament when ejecting into the bowden tube between extruder and hot end, cools and hardens.

### What did you expect to happen

The filament should eject cleanly

### How to reproduce

Fairly obvious.

HelgeKeck, 2024-06-25:
This is not a bug.

RatOS V2.1 has a inbuilt tip forming protocol before it unlaods a filament. it comes with default values that work for msot people, but its not possible to provide settings for every situation, this depends heavily on used nozzle, filament brand and type and selected temperature.

depending you your configuration you need to override the tip forming parameters for your situation, for example if you use a HF CHT nozzle then you override `_UNLOAD_UHF_CHT` in your prionter.cfg

```
[gcode_macro _UNLOAD_UHF_CHT]
gcode:
	# parameter
	{% set toolhead = params.TOOLHEAD|int %}

	DEBUG_ECHO PREFIX="_UNLOAD_UHF_CHT" MSG="TOOLHEAD: {toolhead}"

	_TIP_FORMING RETRACT_LENGTH=18 COOLING_MOVE_LENGTH=10 DIP=False

```

@miklschmidt requesting to close this since its not a bug

smwoodward-beep, 2024-06-25:
Should these not be variables instead of the user needing to override a full macro?

HelgeKeck, 2024-06-25:
> Should these not be variables instead of the user needing to override a full macro?

no, it is handled depending on the used nozzle size and type, there are several of these small macros you can override, this is easier than to provide variables for all different nozzle types

HelgeKeck, 2024-06-25:
if you need more informations i suggest to ask in the discord server

smwoodward-beep, 2024-06-25:
> if you need more informations i suggest to ask in the discord server

Yeah I would love to do that, but I got booted from there because you insulted me telling me I was pissing people off when I told you that over writting ratos.cfg would piss people off.

smwoodward-beep, 2024-06-25:
So until such time as I could use the discord server, I will do that here.

HelgeKeck, 2024-06-25:
i did not insulted you and this is place is purely for bug discussions.

@miklschmidt requesting a close

smwoodward-beep, 2024-06-25:
> i did not insulted you and this is place is purely for bug discussions.
> 
> @miklschmidt requesting a close

This is a bug, and yes you did insult me.

smwoodward-beep, 2024-06-25:
> i did not insulted you and this is place is purely for bug discussions.
> 
> @miklschmidt requesting a close

Now if you would like for me to discuss this in Discord instead of here, let me back in.  Any and all issues I have with RatOS from here forward will be made here otherwise.

closed 2024-06-25 by miklschmidt