Rat-OS/RatOS #141: Nozzle Drag on Rat Rig Logo

dvaldes409, 2024-07-03:
### What happened

On the newest build or two, Using V-core 4 400mm hybrid, print start macro is doing what I'm assuming is the beacon contact probe calibration, and dragging nozzle on the front RatRig logo of the build plate, and then touching the center of the build plate. The force is very small and is not damaging the build plate, but it is transferring green paint into the middle of the print, This is causing a green stain on the surface of the following print. I cannot find the start macro to change it myself for now. Thanks

### What did you expect to happen

Move the y-coordinate of the start print macro front nozzle drag 10mm up.

### How to reproduce

start_print
Did not happen on earlier builds.

### Additional information

RatOS Build v2.1.0-RC1-225-ga5c9d170

HelgeKeck, 2024-07-03:
Hi,

the purpose of this feature is to wipe filament residue from the nozzle tip away. this is the reason it does it on the very front if the build plate

if you like you can turn this feature off by overriding this variable

```
[gcode_macro RatOS]
variable_beacon_contact_wipe_before_calibrate: False     # enables a nozzle wipe at Y0 before doing the contact calibration
```

Helge

dvaldes409, 2024-07-04:
> Hi,
>
> the purpose of this feature is to wipe filament residue from the nozzle tip away. this is the reason it does it on the very front if the build plate
>
> if you like you can turn this feature off [...]

Thanks for the information. It seems like a helpful feature, only issue is that it does it on top of the ratrig logo and carries the green paint onto the center of the build plate. I'll look at the macros to see if I can change the nozzle wipe to the left or right on x-axis so it doesn't touch the logo. Thanks

HelgeKeck, 2024-07-04:
i changed it to make the wipe in the front corner isntead of the front middle

dvaldes409, 2024-07-05:
> i changed it to make the wipe in the front corner isntead of the front middle

Thank you for the quick reply. I just updated to latest update. It looks like the nozzle wipe starts on the front corner now, unfortunately it looks like the end position for the nozzle wipe is still the same, so the nozzle now starts wiping on front left (close to 0,0 ) and continues about 200mm to the middle of the bed before going into the next macro.

Thanks,
David

HelgeKeck, 2024-07-05:
ups

HelgeKeck, 2024-07-05:
fixed it

DavidZisky, 2024-07-30:
@dvaldes409 can you please retest and close the issue if fixed?

dvaldes409, 2024-07-31:
> @dvaldes409 can you please retest and close the issue if fixed?

Absolutely,, it works great. Thanks all.

closed 2024-07-31 by dvaldes409