Rat-OS/RatOS #151: Bed Crashes when at 500mm

malcinator, 2024-07-20:
### What happened

Printed tall object and bed was at 500mm when complete and powered the printer off for the day. The next day I re-homed the printer and tit crashed in to the stepper motors. 

There doesn't seem to be enough room below 500mm to perform the pre-homing bump. Would be good to reduce the pre-homing bump.

### What did you expect to happen

I didn't expect the bed to crash at when at 500mm

### How to reproduce

Lower bed down to 500mm. Power off. Power on and home

miklschmidt, 2024-08-28:
There's no way to prevent this without either risking build plate damage (because you disabled z-hop), reducing your max available z-height or never leaving the bed at 500 before the steppers are disabled. Let the motors skip - it's just electromagnets.

malcinator, 2024-08-28:
I wasn't asking for it to be disabled which would of course risk the build plate but perhaps reducing the z-hop from 15mm to 5mm or even somewhere in-between? Or better still not allowing z-hop when the bed is below 480mm? I understand that the motors skipping is fine in some cases but of a regular basis surely would be damaging and add extra wear to them?

miklschmidt, 2024-08-28:
> Or better still not allowing z-hop when the bed is below 480mm
Klipper doesn't know where the bed is as soon as the steppers are disabled, it doesn't z-hop if the Z is known.

> I understand that the motors skipping is fine in some cases but of a regular basis surely would be damaging and add extra wear to them?
No :)

> I wasn't asking for it to be disabled which would of course risk the build plate but perhaps reducing the z-hop from 15mm to 5mm or even somewhere in-between

If you're comfortable with that, just change it: https://os.ratrig.com/docs/configuration/macros#homing