Rat-OS/RatOS #152: SSID password stored in clear text in ratos-wpa-supplicant.txt

smwoodward-alpha, 2024-07-20:
### What happened

In ratos-wpa-supplicant.txt, the password is commented out but is in clear text in addition to the hashed password

### What did you expect to happen

the wireless password shouldn't be in clear text. The #psk="xxxxx" line should be removed.

### How to reproduce

Looked in ratos-wpa-supplicant.txt and the following lines are there.
ssid="MyWifi"
#psk="printers1234"
psk=xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
scan_freq=5200

funtimeerror, 2024-08-25:
dupe of #142 ?

miklschmidt, 2024-08-28:
@funtimeerror he's a spammer with a personal vendetta, [he created 10 duplicates of this issue](https://github.com/Rat-OS/RatOS/issues?q=is%3Aissue+SSID+password+stored+in+clear+text+in+ratos-wpa-supplicant.txt+is%3Aclosed)

This isn't a bug, it's the direct output of `wpa_passphrase`. Most other pi distributions only use cleartext. However, removing it from the block would be an enhancement, but it has very low priority right now, and it would reduce debuggability. There is still an issue where some special characters don't work correctly. Furthermore, if someone gains access to your pi you have much bigger issues than sharing your wifi password with the intruder. Don't expose your printer to the internet, if you have to, use VPN.

funtimeerror, 2024-09-03:
@miklschmidt should be solvable via regex... i'll take a stab at it in the next week or two

funtimeerror, 2024-09-07:
PR Rat-OS/RatOS-configurator/pull/20 submitted

closed 2024-10-03 by miklschmidt