Rat-OS/RatOS #157: `pin: !fan_part_cooling_pin` method for stopping fan spinning at 100% on boot does not seem to work with my setup

evilC, 2024-10-11:
### What happened

I was following this guide: https://os.ratrig.com/docs/guides/4028/#my-fan-runs-at-100-until-klipper-boots  
I tried the following:
```
[fan]
pin: !fan_part_cooling_pin
cycle_time: 0.00004
enable_pin: PB11
```

While it stopped the fan from spinning at 100% on boot, I lost speed control of the fan (Any setting > 0% speed yielded 100% speed)  

The following did work:
```
[output_pin 4028_POWER]
pin: PB11
value: 1
shutdown_value: 0 # turn the fan power off on MCU shutdown.
```

### What did you expect to happen

Setting should behave as intended

### How to reproduce

Use the supplied entries in printer.cfg

### Additional information

RatOS v2.1.x: v2.1.0-RC2-0
100% stock VCore 4-300 set up as per RR documentation (ie fan PWM connected to EBB toolboard, +ve pin connected to Octopus fan pin)
Only deviation from stock is to change fan -ve pin to Octopus heater pin HE3-

miklschmidt, 2024-10-20:
@evilC you're probably using a toolboard. Just remove the `pin: !fan_part_cooling_pin` part in your override and `enable_pin` should work just fine.

evilC, 2024-10-21:
Confirmed. That fixed it, thanks!

closed 2024-10-21 by evilC