Rat-OS/RatOS #159: Switching from CoreXY to Hybrid removes settings, even if "Use Existing" is selected

evilC, 2024-10-25:
### What happened

I have the following in my printer.cfg to stop the fan spinning at 100% during boot:
```
# Set fan to use heater pin (Fix fan spinning at 100% during boot)
# From https://os.ratrig.com/docs/guides/4028/#my-fan-runs-at-100-until-klipper-boots
[fan]
#pin: !fan_part_cooling_pin  # Do not us with toolboard - see https://github.com/Rat-OS/RatOS/issues/157#issuecomment-2425187293
cycle_time: 0.00004
enable_pin: PB11
```

Whilst trying to debug some layer shift issues, I switched from Hybrid to CoreXY using the configurator
The new printer.cfg completely lacks a `[fan]` section

### What did you expect to happen

It should leave the `[fan]` section alone when switching

### How to reproduce

Enter the setting into printer.cfg, use the configurator to switch from CoreXY to Hybrid

### Additional information

Distro: RatOS v2.1.0-RC 2 (bullseye)

miklschmidt, 2024-12-19:
This is behaving as intended, you explicitly accepted those changes to printer.cfg at the end of the configuration wizard. RatOS cannot resolve the conflicts resulting from user edits of printer.cfg, and changes to printer.cfg are rarely needed, which is why this giant banner is here when you open the diff (which you are required to do to accept changes to printer.cfg):

[image]

miklschmidt, 2024-12-19:
FYI: this will be improved in a future version once it's no longer necessary to write anything to printer.cfg besides the RatOS.cfg include (this requires a configurator interface for variables and limits, etc).

closed 2024-12-19 by miklschmidt