Rat-OS/RatOS #160: "Motor slot conflicts detected" when switching from CoreXY to Hybrid

evilC, 2024-10-27:
### What happened

I have hybrid hardware, and it was initially enabled, but due to doing some debugging I set it to CoreXY mode.
Rather than changing my wiring, instead when I changed to corexy, I selected custom motor ports, such that it uses the same ports as hybrid.
When I went to switch back to Hybrid, I went through the configurator and selected Hybrid.
However, on the "Hardware Selection" page, I got the error `Motor slot conflicts detected`
Furthermore, the Y motors are missing from this page:
[image]

### What did you expect to happen

It should automatically select the default motor configuration

### How to reproduce

Use the RatOS.cfg below in corexy mode and go into the configurator and change to hybrid

### Additional information

This was my RatOS.cfg at the time the error was encountered:
```
# WARNING. THIS FILE IS GENERATED BY THE RATOS CONFIGURATOR.
# CHANGES YOU MAKE HERE WILL BE OVERWRITTEN. KEEP YOUR CHANGES IN PRINTER.CFG.
# Config generated for Rat Rig V-Core 4 300
# Documentation: https://os.ratrig.com

#############################################################################################################
### CONTROLBOARD & TOOLBOARD
#############################################################################################################
[include RatOS/boards/rpi/config.cfg]

[board_pins btt-octopus-11]
aliases:
	#----------------------------------------------- X motor pins -----------------------------------------------
	# Assigned to slot: "MOTOR 0"
	#------------------------------------------------------------------------------------------------------------
	x_step_pin=PF13,
	x_dir_pin=PF12,
	x_enable_pin=PF14,
	x_uart_pin=PC4,
	x_diag_pin=PG6,
	x_endstop_pin=PG6,
	#----------------------------------------------- Y motor pins -----------------------------------------------
	# Assigned to slot: "MOTOR 4"
	#------------------------------------------------------------------------------------------------------------
	y_step_pin=PF9,
	y_dir_pin=PF10,
	y_enable_pin=PG2,
	y_uart_pin=PF2,
	y_diag_pin=PG12,
	y_endstop_pin=PG12,
	#----------------------------------------------- Z motor pins -----------------------------------------------
	# Assigned to slot: "MOTOR 5"
	#------------------------------------------------------------------------------------------------------------
	z0_step_pin=PC13,
	z0_dir_pin=PF0,
	z0_enable_pin=PF1,
	z0_uart_pin=PE4,
	z0_diag_pin=PG13,
	#---------------------------------------------- Z1 motor pins -----------------------------------------------
	# Assigned to slot: "MOTOR 6"
	#------------------------------------------------------------------------------------------------------------
	z1_step_pin=PE2,
	z1_dir_pin=PE3,
	z1_enable_pin=PD4,
	z1_uart_pin=PE1,
	z1_diag_pin=PG14,
	#---------------------------------------------- Z2 motor pins -----------------------------------------------
	# Assigned to slot: "MOTOR 7"
	#------------------------------------------------------------------------------------------------------------
[... 386 lines cut ...]
gcode:
	{% set x = params.X|default(-1.0)|float %}
	{% set y = params.Y|default(-1.0)|float %}
	{% set z = params.Z|default(0.0)|float %}
	{% set s = params.S|default(1)|int %}
	{% if printer["gcode_macro _SELECT_TOOL"] is defined %}
		_SELECT_TOOL T=0 X={x} Y={y} Z={z} TOOLSHIFT={s}
	{% endif %}

# Macro variable overrides
[gcode_macro RatOS]
variable_bed_margin_x: [60, 60]
variable_bed_margin_y: [15, 30]
variable_x_driver_types: ["tmc2209"]
variable_x_axes: ["x"]
variable_y_driver_types: ["tmc2209"]
variable_y_axes: ["y"]
variable_z_driver_types: ["tmc2209", "tmc2209", "tmc2209"]
variable_z_axes: ["z", "z1", "z2"]
variable_home_y_first: True

# Save variables
[save_variables]
filename: /home/pi/printer_data/config/ratos-variables.cfg
```

evilC, 2024-10-28:
I manually fixed this, and all worked OK, but if I subsequently go into the configurator, when I get to the Hardware Selection page, it tries to make changes:
[image]
Furthermore, when I powered up the machine today, it had lost config in RatOS.cfg for stepper X1

miklschmidt, 2024-11-10:
> I manually fixed this, and all worked OK, but if I subsequently go into the configurator, when I get to the Hardware Selection page, it tries to make changes:

What do you mean by "it tries to make changes"?

I think the last problem here is because you didn't confirm your selection at the end of the wizard, any changes you make will not be reflected in your config without explicitly confirming it at the end of the wizard by clicking "Confirm and save":

[image]

It would've been helpful with a debug zip when you had the rendering issue that is depicted in the first post, that's definitely a bug that needs fixing. Your last post seems to show it working as intended.