Rat-OS/RatOS #165: Combination of performance mode and stealthchop mode results in skipped steps

swilkens, 2024-11-10:
### What happened

During automated configuration through the wizzard, select Performance Mode as well as StealthChop mode. The resulting configuration resulted in skipped steps during Z-tilt calibration - which in turn resulted in cracked parts when the lost position caused an unexpected move of the bed.

### What did you expect to happen

A working, safe configurfation

### How to reproduce

1. During configuration wizzard, select Performance Mode as well as StealthChop mode on a standard V3.1 300 x 300 machine. 
2. Initiate homing and Z-Tilt calibration. 
3. Notice missed steps during fast moves (as configured for Z-Tilt).

### Additional information

Discord discussion here:
https://discord.com/channels/582187371529764864/859890291591217162/1305202974875848816

miklschmidt, 2024-11-10:
Fixed by https://github.com/Rat-OS/RatOS-configurator/commit/579428d26f74366ec16f2c3324d5c8e6f532053e

Note you can still select performance mode (which will adjust current of your motors, etc), but the speed limits will be ignored.

miklschmidt, 2024-12-19:
I'm closing this as resolved.

closed 2024-12-19 by miklschmidt