Rat-OS/RatOS #168: Something went wrong...

Adrianjayr, 2024-12-09:
### What happened

I received the following error multiple times at the end of the Balena Etcher validating process. "Something went wrong.  If it is a compressed image, please check that the archive is not corrupted.  The writer process ended unexpectedly."

### What did you expect to happen

Flash file 2023-06-09-RatOS-2.0.2-raspberry-rpi32.img.xz to micro SD

### How to reproduce

I tried with two different SanDisks. The first SanDisk Ultra A1 64GB was purchased from a reputable source.
The second SanDisk Extreme Pro A2 64GB was purchased from a different reputable source.

I tested SanDisks with H2TestW.  H2testw was developed to test USB sticks for various kinds of errors.

I tried both flash formats, Flash from File and Flash from URL.  Please note, Balena Etcher version 1.19.25

### Additional information

None

GP3D84, 2024-12-09:
What is the size of the image downloaded?
there are  know bugs regarding download from GitHub  release or action, in many cases the download was not complete.
eg. RatOS is about 1.2/1.3 GB and it was less than 1gb at the end of download, in this case the file is corrupt.

ATLTVHEAD, 2024-12-09:
I experienced the same error on Windows. I resolved this, by extracting the ". xz" into a folder with the ". img" inside. I then used the img in the folder with etcher and it worked/validated image.

Does that work for you?

Adrianjayr, 2024-12-10:
Thank you for the quick response!  As it turned out, the downloaded file
did not show the zip icon.  I didn't know it was a zipped file.  Please see
attached screenshot.