Rat-OS/RatOS #178: 502 Bad getway after update RatOS core packages

hainguyents13, 2025-01-17:
### What happened

Error 502 Bad Getway when access http://ratos.local/configure after updating ratos core packages

### What did you expect to happen

I want to try new ratos version, so I downloaded https://github.com/Rat-OS/RatOS/releases/tag/v2.1.0-RC2 then flash to my SD card, BelenaEtcher not working so I use rufus and it worked.
After config wifi, I follow the config steps here https://os.ratrig.com/docs/installation/#updating-ratos by updating ratos core packages, then when I continue to the confiurator, the page shows 502 Bad Getway error

### How to reproduce

- Download and write RatOS v2.1.0-RC2
- Config wifi
- Update RatOS core packages
   1. Update `RatOS`
   2. Update `ratos-configurator`
- Go to http://ratos.local/configure

Elfstennr1, 2025-01-25:
Same problem for me. 

I could install if I skipped the part with updating before continuing, but efter printer vas setup and working I updated and the system collapsed with plenty of errors.

greatguitartist, 2025-01-26:
Same here. I updated RatOS core packages and now I am unable to access configurator and I also get a bunch of moonraker warnings. I tried to manually install RatOS configurator using the command git clone git@github.com:Rat-OS/RatOS-configurator.git, thinking perhaps if I reinstalled the package and ran the install script, it may fix the issue, but I get access denied errors. 

[image]

[image]

[image]

greatguitartist, 2025-01-27:
> Same here. I updated RatOS core packages and now I am unable to access configurator and I also get a bunch of moonraker warnings. I tried to manually install RatOS configurator using the command git clone [git@github.com](mailto:git@github.com):Rat-OS/RatOS-configurator.git, thinking perhaps if I reinstalled the package and ran the install script, it may fix the issue, but I get access denied errors.
> 
> [image]
> 
> [image]
> 
> [image]
> 
> [image]

I no longer have this issue. I reflashed RatOS 2.1 on my SD Card and did all the setup before updating the packages and the printer is now fine. Still an issue if you follow the documentation steps, but workaround exists.

Pepe-FM, 2025-01-28:
Same problem here. Updated to latest packages, and klipper throws error connecting to mcu, lots of moonraker warnings and bad gateway on configurator.
Tried reinstalling ratOS 2.1 again. Everything working fine if I don't update to latest configurator.

LordMM74, 2025-03-16:
Are there any news here?
When will this be fixed? Which version of configurator works?

JayPerez1, 2025-04-02:
Same here. I started a brand-new install on a BTT C1 Pi V1.2 and using the latest pre-release RatOS-v2.1.0-RC2

hainguyents13, 2025-04-16:
For anyone facing this issue, try rewriting the OS using older version of Balena, I tried to use v1.8 and it works fine now, the problem likely comes from how it was written to the sd card

closed 2025-04-16 by hainguyents13

petjek, 2025-04-23:
> For anyone facing this issue, try rewriting the OS using older version of Balena, I tried to use v1.8 and it works fine now, the problem likely comes from how it was written to the sd card

I don't think so. I had some trouble with Balena too and switched to Pi Imager to write the image. Worked fine until I installed the latest OS updates. Configurator is v2.1.0-RC3-83-g3a6a8b01