Rat-OS/RatOS #179: Unable to home after latest klipper update

greatguitartist, 2025-01-24:
### What happened

I updated the Klipper service on my V-Core 3.1 running RatOS 2.0 and since then, when I try to home the printer, I get an error message "Must home axis first 0.000 0.000 7.000 [0.000]. This error message pops up no matter whether I try to home an individual axis or all axes. The only way I am currently able to home the printer is by first running the command SET_CENTER_KINEMATIC_POSITION and then running the home commands.

### What did you expect to happen

The printer should home without errors without having to run SET_CENTER_KINEMATIC_POSITION

### How to reproduce

Update Klipper to the latest commits, restart the printer, and try to home the printer (try a single axis or all axes). When the error is received, run macro SET_CENTER_KINEMATIC_POSITION and then try to home.

### Additional information

Klippy Log: 

[klippy.log](https://github.com/user-attachments/files/18541327/klippy.log)

Screenshot of error message:

[image]

greatguitartist, 2025-01-24:
I used KIAUH to roll back to a stable version of Klipper and things are working now. The commits that I rolled back are. I should not need to do this on my printer, so please let me know how this can be fixed without working around Klipper commits.

[image]

greatguitartist, 2025-01-27:
Upgraded to RatOS2.1-RC3, so this issue no longer applies.

closed 2025-01-27 by greatguitartist

seb449, 2025-02-03:
Hello,
you have a solution ?
Thank

greatguitartist, 2025-02-03:
I upgraded to RatOS 2.1, so this issue is no longer open.

robertweis56, 2025-02-07:
Hi, I wonder where you find RatOS2.1-RC3. I only find RC2

greatguitartist, 2025-02-07:
I think that was a typo. My bad. I upgraded to 2.1-RC2

robertweis56, 2025-02-07:
Ok, thanks for answering

Idire, 2025-04-26:
Hello.
I upgraded recently to 2.1-RC3 (precisely `v2.1.0-RC3-26-g33980992`) and I am experiencing exactly the same problem.

Krki012, 2025-05-08:
I have the latest version and yet still have the issue of not being able to home

andreab3, 2025-07-28:
Looks to me like klippy/extras/ratos_homing.py tries to move on Z axis, but that isn't allowed if X/Y aren't homed.  Setting z_hop to zero prevents this and allowed me to home X and Y axis:

[ratos_homing]
z_hop: 0

dnehring, 2025-08-14:
> Looks to me like klippy/extras/ratos_homing.py tries to move on Z axis, but that isn't allowed if X/Y aren't homed. Setting z_hop to zero prevents this and allowed me to home X and Y axis:
> 
> [ratos_homing] z_hop: 0

Thanks for this suggestion, this also resolved my issue when updating to 2.1-RC3