Rat-OS/RatOS #187: RatOs 2.0.2 Updated now G28 error killing Klipper

smbenson0712, 2025-04-07:
### What happened

Updated RatOs on V-Minion and try to home the machine and I get Internal error on command:"G28".  On the update manager there is an information waring for klipper - Repo has untracked source files: ['klippy/extras/beacon.py', 'klippy/extras/gcode_shell_command.py', 'klippy/extras/linear_movement_vibrations.py', 'klippy/extras/ratos_homing.py']
From Klippy.log file:
Internal error on command:"G28"
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/gcode.py", line 212, in _process_commands
    handler(gcmd)
  File "/home/pi/klipper/klippy/extras/ratos_homing.py", line 42, in cmd_G28
    toolhead.set_position(pos, homing_axes=[2])
  File "/home/pi/klipper/klippy/toolhead.py", line 466, in set_position
    self.kin.set_position(newpos, homing_axes)
  File "/home/pi/klipper/klippy/kinematics/cartesian.py", line 69, in set_position
    axis = "xyz".index(axis_name)
TypeError: must be str, not int
Transition to shutdown state: Internal error on command:"G28"
Looks like sending a INT and not xy or z to the home axes.

### What did you expect to happen

Home machine as normal.  How do I fix this error?

### How to reproduce

Not sure.

### Additional information

klipper
v0.12.0-468-gd679f711
RatOS
v2.0.2-36-g230929e3

proto4type, 2025-04-07:
I Reinstalled RatOS 2.02 fresh from github.  Copied back my printer.cfg and was required to update klipper as the pi klipper version was 0.11 on the fresh install and the octopus mcu was at 0.12 version of klipper.  Updated RatOs first and then updated all components of the install.  After klipper was updated along with all other components, mainsail loaded normally.  Homing the machine gives the same error, so updating to the latest klipper version breaks the homing code above.

smbenson0712, 2025-04-07:
Reproduced with fresh install and update.
Reinstalled RatOS 2.02 fresh from github on to the sd card. Copied back my printer.cfg and was required to update klipper as the pi klipper version was 0.11 on the fresh install and the octopus mcu was at 0.12 version of klipper. Updated RatOs first and then updated all components of the install. After klipper was updated along with all other components, mainsail loaded normally. Homing the machine gives the same error, so updating to the latest klipper version breaks the homing code above.

markleaf131313, 2025-04-19:
Yah doesn't seem like they are maintaining 2.0 anymore even when 2.1 isn't released yet. This has been reported in a few different places over the last few months.

The problem is their custom homing code, it moves the bed down before homing and I guess that klipper changed something that makes it fail.

https://github.com/Rat-OS/RatOS-configuration/blob/3a6a8b018fd46ef4343a246a66b8c0fc357f05a4/klippy/ratos_homing.py#L32

Quick fix to get it working is to set z_hop to 0 in ratos_homing, but this means it doesn't lower the bed before homing and after homing so the nozzle moves to z = 0.

smbenson0712, 2025-04-19:
Rolled back with git to an older version of klipper to make it work.  Just won't update unless they fix it.

bzed, 2025-05-30:
> Quick fix to get it working is to set z_hop to 0 in ratos_homing, but this means it doesn't lower the bed before homing and after homing so the nozzle moves to z = 0.

The proper quickfix is to use

`toolhead.set_position(pos, homing_axes='z')` int

https://github.com/Rat-OS/RatOS-configuration/blob/3a6a8b018fd46ef4343a246a66b8c0fc357f05a4/klippy/ratos_homing.py#L42

miklschmidt, 2025-06-06:
Indeed 2.x is not going to see much work. 2.1 is already far ahead of where it ever was, so it doesn't seem to make much sense. It'll be out of RC soon™.

ignacmc, 2025-06-11:
I have found the same problem in RatOS 2.1.0 RC3 and I have solved it with the help of DeepSeek  ;-)

### Fix That Worked for Me  
Changed line 42 in `/home/pi/klipper/klippy/extras/ratos_homing.py` from:  
```python
toolhead.set_position(pos, homing_axes=[2])  # ❌ Broken
to:

python
toolhead.set_position(pos, homing_axes=['z'])  # ✅ Fixed
After a Klipper restart (sudo service klipper restart), homing works perfectly! 🎉

Why This Matters
This broke my printer completely until I patched it

Several users on forums have similar issues (but no solutions posted)

Likely affects all RatOS 2.1 RC3 + Klipper v0.13 setups

Suggested Permanent Fix
Update ratos_homing.py to use axis letters instead of numbers. Simple patch:

diff
- toolhead.set_position(pos, homing_axes=[2])
+ toolhead.set_position(pos, homing_axes=['z'])
My Setup
Printer: RatRig V-Minion

Board: BTT Octopus 1.1 (F446)

OS: RatOS 2.1 RC3

Klipper: v0.13

Hope this helps others! This bug is a real showstopper 😅
Keep up the great work otherwise – loving RatOS!

It

miklschmidt, 2025-07-03:
@ignacmc RC3 pins klipper, so you'll manually have to move beyond the pinned commit to break it, it doesn't happen by itself from following the instructions and using moonraker.