Rat-OS/RatOS #191: Process of installing the new 2.1.0 version

Krki012, 2025-05-03:
### What happened

For anyone reading through the issues on how to make ratOS run, I am writing a small step list that made my setup run so it might help you out. 

My image that I flashed to the SD card is 2024-05-28-RatOS-2.1.0-RC1-raspberry-rpi32.img.xz 

**First step:** 
I'll assume you already flashed the SD card, put it in your raspberry or similar device and configured the wifi/ethernet and hostname. Everywhere where I speak of RatOS as hostname, make sure to change it to your own hostname you set up after connecting to the internet.

Now, before doing eanything else with the image, go to terminal, ssh into your pi with 
`ssh pi@ratos.local` 
with "raspberry" as a password and do the following:
`sudo apt update` and `sudo apt upgrade` in that order.

**Second step:**
Once that's done, we have to update all the components on the ratos.local/config tab. Scroll down and check whether there is any of the components that say "invalid". If that's the case, update the RatOS (if available) and then  just do a hard recovery which should reinstall it and possibly update the component. If the RatOS is the invalid one, do it first before any other. For me it was always available after update and upgrade.

**Third step:**
Now that the components are installed and updated, you should get moonraker error that setup.sh file doesn't exist. 
("[update_manager]: Failed to load extension ratos-configurator: Invalid path for option `install_script` in section [update_manager ratos-configurator]: Path `/home/pi/ratos-configurator/scripts/setup.sh` does not exist.")

To fix this, ssh once again to raspbery pi (if necessary) and paste this command:
`cp /home/pi/ratos-configurator/app/scripts/setup.sh /home/pi/ratos-configurator/scripts`
which will copy the setup.sh file from app folder to the location where moonraker expects to find it (scripts folder).
Reboot and the issue should go away.

**Fourth step:**
Changing to the configurator tab should result in error 502: bad gateway. SSH in to the pi. Running an update script manually seems to fix the issue. You can do:
`cd /home/pi/ratos-configurator/`
`sudo app/scripts/update.sh`
Password is again "raspberry". After this step, you might need to update klipper once again.

**Lastly** you will see an error on dashboard about TEST_RESONANCES. This issue's solution can be found [here](https://www.answeroverflow.com/m/1323218447924199486), but that just requires you to continue with the setup wizard on ratos.local/configure/wizard and the issue should be solved by assigning an accelerometers to x and y axis. 

This is as far as I got for now, I'll update upon this after I get to the office to the printer. I'm building an sd card setup at home and finishing it in the office.

### What did you expect to happen

I expected this outcome as it's not yet in full release

### How to reproduce

Just follow their guide and you'll get those issues

Krki012, 2025-05-03:
I'll put this as completed once I verify it's operational. This is for anyone having issues and to show the developers where issues may arrise

Krki012, 2025-05-06:
I've started to work on them and so far, I only had issues with selecting the right thermistor. IDK why almost every thermistor reports temperature drop when it's in fact rising. 

Also, another issue I found is that once I save the config, raspberry reboots for some reason. I'll look more into it and update

Krki012, 2025-05-15:
Update:

Thermistors were PT1000 but the connections to board were bad. 

Part cooling fan is connected to "Input voltage PWM" on the toolboard but needs to be inverted in RatOS.cfg. Hotend fan needs to be inverted too.

Home all needs to have the probe (beacon in my case) to be calibrated first so the order is:

1. Home X `G28 X`
2. Home Y `G28 Y`
3. `BEACON_CALIBRATE`
4. `SAVE_CONFIG`
5. Z-TILT calibration `Z_TILT_ADJUST`
     -If the issue is that the difference in calibration is increasing, you probably reversed the front two motors (like me)

matthewj301, 2025-05-15:
I would mention this in the ratos-development channel in the RatOS discord, This seems like a lot of manual steps and issues that I personally did not have to do or run into (have it RatOS running on 3 machines - V-Core 3.1, Doomcube, tri-zero)

Krki012, 2025-05-16:
I will relay it further. Thanks

closed 2025-05-21 by Krki012