Rat-OS/RatOS #193: Need homing before homing

Krki012, 2025-05-15:
### What happened

I was tasked with installing RatOS on 4 new V-Core 4 500 printers and after flashing all cards and setting up all of them as described in my issue #191, printer number 2 presented me with an issue when trying to home: "Must home axis first: 0.000 0.000 15.000 [0.000]". 

Now, I can bypass that by using SET_CENTER_KINEMATIC_POSITION and then homing, but I'd like to avoid doing that every time before printing or servicing the printer. 

I've attached the klippy and moonraker logs for you to check out. I really have no more ideas on how to fix this. I spent hours comparing the cfg files (all of them) and they are all identical but printer number 2 is acting up. Anyone knows what the issue might be? I've already checked the homing.cfg

[klippy.log](https://github.com/user-attachments/files/20226734/klippy.log)

[moonraker.log](https://github.com/user-attachments/files/20226738/moonraker.log)

### What did you expect to happen

Didn't expect it to ask for homing before homing for sure

### How to reproduce

Idk, it happened on one of four that I was installing

matthewj301, 2025-05-15:
Does this happen when you press "home", or when you initiate a print and it expects to be in a "homed" state but is unhomed?

matthewj301, 2025-05-15:
If this happens when initiating a print, can you paste you start gcode from your slicer?

Krki012, 2025-05-15:
This happens when I hit home all button. We still didn't get the plates to print on so I wouldn't know to answer if it's present when starting print

matthewj301, 2025-05-15:
No worries, just trying to nail down context, I appreciate that info! Just double checking, you installed all updates and everything?

Krki012, 2025-05-16:
I have tried using the G28 directly as well as G28 X and G28 Y. BEACON_CALIBRATE requires homing of X and Y before proceeding. They all resulted in the same message

Krki012, 2025-05-16:
I checked all the cfg files and they are identical (all of them, even the ones in RatOS folder). I even tried checking the javascript to find where the buttons lead me and what they do but it didn't really help

matthewj301, 2025-05-21:
At this point I would reflash the microsd and try to go through the official RatOS setup docs

Krki012, 2025-05-21:
I did that and the issue seems to gave gone away. Strange nonetheless

closed 2025-05-21 by Krki012