Rat-OS/RatOS #194: [fluidd] Timelapse tab not showing up

depau, 2025-05-15:
### What happened

(opening up here since the fluidd repo doesn't have an issue tracker - I'm using the fluidd fork, although I think the issue happens with mainsail as well)

I enabled a webcam and timelapses in moonraker and it works well, however the tab does not display on the side bar. If I manually navigate to `/#/timelapse` everything works fine. The history tab also doesn't appear on the sidebar but works if invoked manually.

### What did you expect to happen

I'd expect the tabs to show up.

### How to reproduce

1. Configure a webcam in crowsnest

```ini
[crowsnest]
log_path: /home/pi/printer_data/logs/crowsnest.log
log_level: verbose
delete_log: false
no_proxy: false‍

[cam vaoc]
mode: camera-streamer
enable_rtsp: false
rtsp_port: 8554
port: 8080
device: /dev/v4l/by-id/usb-3DO_Rat_Rig_VAOC_Camera_3DO-video-index0
resolution: 1920x1080
max_fps: 30

[cam enclosure]
mode: camera-streamer
enable_rtsp: false
rtsp_port: 8555
port: 8081
device: /dev/v4l/by-id/usb-3DO_3DO_USB_CAMERA_V2_3DO-video-index0
resolution: 1920x1080
max_fps: 30
v4l2ctl: autoexposure=1,exposuretimeabsolute=2000,powerlinefrequency=1
```

2. Enable timelapse in Moonraker

```ini
[timelapse]
# empty
```

3. Configure the camera and the timelapse via Fluidd
4. Observe how the timelapse tab doesn't show up.

### Additional information

As I [mentioned on Discord](https://discord.com/channels/582187371529764864/1276343953452564512/1372261456741142580) I checked with the Vue DevTools and it looks like the state value that is used to check whether the tabs should be enabled seems to have the correct values, however it's still not working.

[image]

miklschmidt, 2025-06-06:
I don't know whether @HelgeKeck is still interested in supporting that fork, if not it's unlikely to get further updates until someone else gets involved. I don't have the resources to take it on myself unfortunately.

HelgeKeck, 2025-06-07:
i do not have the resources to continue to work on this fork. im sorry. its still my favorite UI but its pretty much dead unless someone wants to do that. my apologies

miklschmidt, 2025-06-08:
@HelgeKeck no worries my man, spare time is precious!

depau, 2025-06-08:
I understand, no worries. The workaround is good enough for me, I'll stick with that or switch back to Mainsail. Take care!

HelgeKeck, 2025-06-08:
> I understand, no worries. The workaround is good enough for me, I'll stick with that or switch back to Mainsail. Take care!

you can have a look at mainline fluidd, pedro implemented many of the features from that fork

HelgeKeck, 2025-06-09:
> [@HelgeKeck](https://github.com/HelgeKeck) should i switch the `fluidd-experimental` option to mainline fluidd instead?

good idea.